FreeCAD/FreeCAD #20743: Null shape on 3D offset of pad made from imported DXF

dyfrgi, 2025-04-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I imported a DXF of the 20-2040 profile from 8020.net (downloaded from https://8020.net/downloads/index/designfiles/). I then went into Draft and used Draft to sketch to convert all of the imported lines and curves into a sketch. Then I used the Sketcher workbench to correct the geometry with Validate sketch, Missing Coincidences -> Find, Missing Coincidences -> Fix. This fixed 184 missing coincidences. After this, none of the options in Sketcher validation find issues.

Then I went into the Part workbench and did an Extrude operation to create a solid from the sketch. This solid looks correct. It also passes Part -> Check Geometry, with no issues found. I also checked some dimensions using the Measure tool and they are as expected from the diagrams provided by 8020.net.

Next I wanted to create an offset version of this object to use as a subtraction operation on another solid so as to make a shape that would fit into this extrusion that can be 3D printed. Now I get an error:
```
<Exception> TopoShapeExpansion.cpp(1395): Null shape
Offset: Null shape
```

[image]

The files may be found at https://github.com/dyfrgi/2040-adapters.

### Full version info

```shell
OS: NixOS 25.05 (Warbler) (niri/niri/wayland)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.11, Qt 5.15.16, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * workfeature-macro
  * Dracula 0.0.6
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dyfrgi, 2025-04-10:
With a bit more experimentation, it is capable of doing an offset if I set the offset amount low enough. 0.15mm for instance. Perhaps this is just an issue of a confusing error message. Fortunately I think this is enough for my purposes so I shouldn't have to figure out how to do it better.

Draft's offset operator chokes on this file, while Sketch is okay-ish though of course it's very unhappy about it being thoroughly unconstrained. So an alternative option for someone else doing this might be to try it with Sketch.

mstodd, 2025-05-23:
I have a similar issue. My sketch is fully constrained and a .2mm offset works if one arc has a radius of 2.65mm, but not 2.6mm.
If I change my offset to .1mm I can then reduce my radius below 2.5mm.
The sketch is fully constrained and fine, the extrusion is fine and passes geometry check, but there's something wrong with the offset though visually I can't tell what it would be.

FEA-eng, 2026-09-26:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```