FreeCAD/FreeCAD #20750: Part Design, Chamfer on adjacent pads fails: Chamfer: BRep_API: command not done

kweb-, 2025-04-10:
[image]

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/69.9239/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.59
  * fasteners 0.5.37
  * POV-Ray-Rendering
  * sheetmetal 0.7.21

Macro below creates doc, two pads from sketches, and attempts to chamfer inside top edge of pads.  This fails with "Chamfer: BRep_API: command not done" dialog. The only way to complete is to chamfer one edge at 3mm and select the other edge and the small corner line segment and set chamfer to 2.9mm for both.  

```py
import FreeCAD
import PartDesign
import PartDesignGui
import Sketcher

doc = FreeCAD.newDocument('sonOfChamferBug')

App.activeDocument().addObject('PartDesign::Body','Body')
App.ActiveDocument.getObject('Body').Label = 'Body'

App.ActiveDocument.recompute()

App.ActiveDocument.getObject('Body').newObject('Sketcher::SketchObject','Sketch')
App.ActiveDocument.getObject('Sketch').AttachmentSupport = (App.ActiveDocument.getObject('XZ_Plane'),[''])
App.ActiveDocument.getObject('Sketch').MapMode = 'FlatFace'
App.ActiveDocument.recompute()

ActiveSketch = App.ActiveDocument.getObject('Sketch')

lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.LineSegment(App.Vector(0.000000, 0.000000, 0.000000),App.Vector(0.000000, 50.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(0.000000, 50.000000, 0.000000),App.Vector(-50.000000, 50.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(-50.000000, 50.000000, 0.000000),App.Vector(-50.000000, 0.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(-50.000000, 0.000000, 0.000000),App.Vector(0.000000, 0.000000, 0.000000)))
App.ActiveDocument.getObject('Sketch').addGeometry(geoList,False)
del geoList

constraintList = []
constraintList.append(Sketcher.Constraint('Coincident', 0, 2, 1, 1))
constraintList.append(Sketcher.Constraint('Coincident', 1, 2, 2, 1))
constraintList.append(Sketcher.Constraint('Coincident', 2, 2, 3, 1))
constraintList.append(Sketcher.Constraint('Coincident', 3, 2, 0, 1))
constraintList.append(Sketcher.Constraint('Vertical', 0))
constraintList.append(Sketcher.Constraint('Vertical', 2))
constraintList.append(Sketcher.Constraint('Horizontal', 1))
constraintList.append(Sketcher.Constraint('Horizontal', 3))
App.ActiveDocument.getObject('Sketch').addConstraint(constraintList)
del constraintList

App.ActiveDocument.getObject('Sketch').addConstraint(Sketcher.Constraint('Distance',0,1,2,2,50.000000)) 
App.ActiveDocument.getObject('Sketch').addConstraint(Sketcher.Constraint('Distance',1,1,3,2,50.000000)) 
App.ActiveDocument.getObject('Sketch').addConstraint(Sketcher.Constraint('Coincident', 0, 1, -1, 1))

App.ActiveDocument.recompute()

App.ActiveDocument.recompute()

App.getDocument('sonOfChamferBug').recompute()

App.ActiveDocument.getObject('Body').newObject('PartDesign::Pad','Pad')
App.ActiveDocument.getObject('Pad').Profile = (App.ActiveDocument.getObject('Sketch'), ['',])
App.ActiveDocument.getObject('Pad').Length = 10
App.ActiveDocument.recompute()
App.ActiveDocument.getObject('Pad').ReferenceAxis = (App.ActiveDocument.getObject('Sketch'),['N_Axis'])
App.ActiveDocument.getObject('Sketch').Visibility = False
App.ActiveDocument.recompute()

App.ActiveDocument.getObject('Pad').Length = 4.500000
App.ActiveDocument.getObject('Pad').TaperAngle = 0.000000
App.ActiveDocument.getObject('Pad').UseCustomVector = 0
App.ActiveDocument.getObject('Pad').Direction = (0, -1, 0)
App.ActiveDocument.getObject('Pad').ReferenceAxis = (App.ActiveDocument.getObject('Sketch'), ['N_Axis'])
App.ActiveDocument.getObject('Pad').AlongSketchNormal = 1
App.ActiveDocument.getObject('Pad').Type = 0
App.ActiveDocument.getObject('Pad').UpToFace = None
App.ActiveDocument.getObject('Pad').Reversed = 0
App.ActiveDocument.getObject('Pad').Midplane = 0
App.ActiveDocument.getObject('Pad').Offset = 0
App.getDocument('sonOfChamferBug').recompute()

App.ActiveDocument.getObject('Sketch').Visibility = False

App.ActiveDocument.getObject('Body').newObject('Sketcher::SketchObject','Sketch001')
App.ActiveDocument.getObject('Sketch001').AttachmentSupport = (App.ActiveDocument.getObject('YZ_Plane'),[''])
App.ActiveDocument.getObject('Sketch001').MapMode = 'FlatFace'
App.ActiveDocument.recompute()

App.ActiveDocument.getObject('Sketch001').addExternal("Pad","Edge3")
App.ActiveDocument.recompute()

ActiveSketch = App.ActiveDocument.getObject('Sketch001')

lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.LineSegment(App.Vector(0.000000, 50.000000, 0.000000),App.Vector(0.000000, 0.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(0.000000, 0.000000, 0.000000),App.Vector(50.000000, 0.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(50.000000, 0.000000, 0.000000),App.Vector(50.000000, 50.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(50.000000, 50.000000, 0.000000),App.Vector(0.000000, 50.000000, 0.000000)))
App.ActiveDocument.getObject('Sketch001').addGeometry(geoList,False)
del geoList

constraintList = []
constraintList.append(Sketcher.Constraint('Coincident', 0, 2, 1, 1))
constraintList.append(Sketcher.Constraint('Coincident', 1, 2, 2, 1))
constraintList.append(Sketcher.Constraint('Coincident', 2, 2, 3, 1))
constraintList.append(Sketcher.Constraint('Coincident', 3, 2, 0, 1))
constraintList.append(Sketcher.Constraint('Vertical', 0))
constraintList.append(Sketcher.Constraint('Vertical', 2))
constraintList.append(Sketcher.Constraint('Horizontal', 1))
constraintList.append(Sketcher.Constraint('Horizontal', 3))
App.ActiveDocument.getObject('Sketch001').addConstraint(constraintList)
del constraintList

App.ActiveDocument.getObject('Sketch001').addConstraint(Sketcher.Constraint('Distance',0,1,2,2,50.000000)) 
constraintList = []
constraintList.append(Sketcher.Constraint('Coincident', 0, 1, -3, 2))
constraintList.append(Sketcher.Constraint('PointOnObject', 1, 2, -1))
App.ActiveDocument.getObject('Sketch001').addConstraint(constraintList)
del constraintList

App.ActiveDocument.recompute()

App.ActiveDocument.recompute()

App.getDocument('sonOfChamferBug').recompute()

App.ActiveDocument.getObject('Body').newObject('PartDesign::Pad','Pad001')
App.ActiveDocument.getObject('Pad001').Profile = (App.ActiveDocument.getObject('Sketch001'), ['',])
App.ActiveDocument.getObject('Pad001').Length = 10
App.ActiveDocument.recompute()
App.ActiveDocument.getObject('Pad001').ReferenceAxis = (App.ActiveDocument.getObject('Sketch001'),['N_Axis'])
App.ActiveDocument.getObject('Sketch001').Visibility = False
App.ActiveDocument.recompute()

App.ActiveDocument.getObject('Pad001').Length = 4.500000
App.ActiveDocument.getObject('Pad001').TaperAngle = 0.000000
App.ActiveDocument.getObject('Pad001').UseCustomVector = 0
App.ActiveDocument.getObject('Pad001').Direction = (1, 0, 0)
App.ActiveDocument.getObject('Pad001').ReferenceAxis = (App.ActiveDocument.getObject('Sketch001'), ['N_Axis'])
App.ActiveDocument.getObject('Pad001').AlongSketchNormal = 1
App.ActiveDocument.getObject('Pad001').Type = 0
App.ActiveDocument.getObject('Pad001').UpToFace = None
App.ActiveDocument.getObject('Pad001').Reversed = 1
App.ActiveDocument.getObject('Pad001').Midplane = 0
App.ActiveDocument.getObject('Pad001').Offset = 0
App.getDocument('sonOfChamferBug').recompute()
App.ActiveDocument.getObject('Pad').Visibility = False

App.ActiveDocument.getObject('Sketch001').Visibility = False

App.ActiveDocument.getObject('Body').newObject('PartDesign::Chamfer','Chamfer')
App.ActiveDocument.getObject('Chamfer').Base = (App.ActiveDocument.getObject('Pad001'),['Edge22','Edge11',])

App.ActiveDocument.getObject('Pad001').Visibility = False
App.ActiveDocument.recompute()

App.ActiveDocument.getObject('Chamfer').Size = 3.000000
App.ActiveDocument.getObject('Chamfer').Base = (App.ActiveDocument.getObject('Pad001'),["Edge22","Edge11",])
App.getDocument('sonOfChamferBug').recompute()

```

hyarion, 2025-04-10:
I’ve transferred the issue to the correct repository

FEA-eng, 2025-12-14:
See [this](https://github.com/FreeCAD/FreeCAD/issues/20750#issuecomment-3378051521)

FEA-eng, 2026-09-14:
[This](https://github.com/FreeCAD/FreeCAD/issues/20750#issuecomment-3378051521) is still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 616.92/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```