FreeCAD/FreeCAD #20752: PPA: daily build failure 2025-03-25

mfraser, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The PPA daily builds have been failing for about a week, since 2025-03-25 [freecad-daily Builds](https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+packages?field.name_filter=&field.status_filter=&field.series_filter=noble). This is for 2404, but 2004 & 2204 are also failing.

Looking at the [build log](https://launchpadlibrarian.net/787536795/buildlog_ubuntu-noble-amd64.freecad-daily_1.1~pre1~202504110102~ubuntu24.04.1_BUILDING.txt.gz)

It fails at this point.
```
dh_auto_configure: error: cd debian/build-py3 && DEB_PYTHON_INSTALL_LAYOUT=deb cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON -DFETCHCONTENT_FULLY_DISCONNECTED=ON -DCMAKE_INSTALL_RUNSTATEDIR=/run "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DBUILD_FLAT_MESH=ON -DFREECAD_BUILD_DEBIAN=ON -DFREECAD_USE_PYBIND11=ON "-DCMAKE_CXX_FLAGS=-g -O2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/<<PKGBUILDDIR>>=. -flto=auto -ffat-lto-objects -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -fdebug-prefix-map=/<<PKGBUILDDIR>>=/usr/src/freecad-daily-1.1~pre1~202504110102~ubuntu24.04.1 -Wdate-time -D_FORTIFY_SOURCE=3 -Wall -fpermissive" "-DCMAKE_C_FLAGS=-g -O2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/<<PKGBUILDDIR>>=. -flto=auto -ffat-lto-objects -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -fdebug-prefix-map=/<<PKGBUILDDIR>>=/usr/src/freecad-daily-1.1~pre1~202504110102~ubuntu24.04.1 -Wdate-time -D_FORTIFY_SOURCE=3 -Wall -fpermissive" "-DCMAKE_SHARED_LINKER_FLAGS=-Wl,-Bsymbolic-functions -flto=auto -ffat-lto-objects -Wl,-z,relro -Wl,-z,now -Wl,--as-needed" -DLIB_SUFFIX= -DENABLE_DEVELOPER_TESTS=OFF -DINSTALL_TO_SITEPACKAGES=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_DATADIR=/usr/share/freecad-daily -DCMAKE_INSTALL_DOCDIR=/usr/share/doc/freecad-daily-doc -DCMAKE_INSTALL_PREFIX=/usr/share/freecad-daily "-DCMAKE_CXX_FLAGS=-g -O2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/<<PKGBUILDDIR>>=. -flto=auto -ffat-lto-objects -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -fdebug-prefix-map=/<<PKGBUILDDIR>>=/usr/src/freecad-daily-1.1~pre1~202504110102~ubuntu24.04.1 -Wdate-time -D_FORTIFY_SOURCE=3 -Wall -fpermissive -I/usr/include/python3.12" -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIRS=/usr/include/python3.12 -DCMAKE_INSTALL_LIBDIR=/usr/lib/freecad-daily-python3/lib ../.. returned exit code 1
make[1]: *** [debian/rules:39: override_dh_auto_configure] Error 2
make[1]: Leaving directory '/<<PKGBUILDDIR>>'
make: *** [debian/rules:3: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git)
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 100.911/102.706/1
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.6.2
  * Gridfinity 0.11.6
  * fasteners 0.5.38
  * dodo 1.0.1
  * Curves 0.6.61
  * ThreadProfile 1.93.0
  * sheetmetal 0.7.22
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * DynamicData.backup1738429910.4145672 2.70.0 (Disabled)
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.8
  * DynamicData 2.73.0
  * FreeCAD-themes 2025.1.7
  * Defeaturing 1.2.4
  * Curves.backup1725432442.4530568 0.6.46 (Disabled)
  * CurvedShapes 1.0.13
  * lattice2.backup1738429917.0355425 1.0.0 (Disabled)
  * ThreadProfile.backup1741800465.9496617 1.92.0 (Disabled)
  * Assembly4 0.12.4
  * freecad.gears 1.3.0
  * A2plus 0.4.68
  * fcgear 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-11:
Thanks for filing the bug report. I believe it's a duplicate of https://github.com/FreeCAD/FreeCAD-Bundle/issues/397

furgo16, 2025-04-11:
The actual issue in the log is:

```
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter Development Development.Module Development.Embed 
CMake Error at cMake/FreeCAD_Helpers/SetupPCL.cmake:11 (find_package):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" with any of
  the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:82 (SetupPCL)

-- Configuring incomplete, errors occurred!
	cd debian/build-py3 && tail -v -n \+0 CMakeCache.txt
```

The Debian source package in the PPA needs to add the `libpcl-dev` dependency at the very least.