FreeCAD/FreeCAD #20753: Possible error in SketchObject::getHigherElements

hyarion, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The indentation is off which hides that half the function is deadcode due to what I think is a wrongly placed return:

https://github.com/FreeCAD/FreeCAD/blame/main/src/Mod/Sketcher/App/SketchObject.cpp#L11537-L11577
```cpp
std::vector<Data::IndexedName>
SketchObject::getHigherElements(const char *element, bool silent) const
{
    std::vector<Data::IndexedName> res;
        if (boost::istarts_with(element, "vertex")) {  // <-- notice how this indented
            int n = 0;
            int index = atoi(element+6);
            for (auto cstr : Constraints.getValues()) {
                ++n;
                if (cstr->Type != Sketcher::Coincident)
                    continue;
                if(cstr->First >= 0 && index == getSolvedSketch().getPointId(cstr->First, cstr->FirstPos) + 1)
                    res.push_back(Data::IndexedName::fromConst("Constraint", n));
                if(cstr->Second >= 0 && index == getSolvedSketch().getPointId(cstr->Second, cstr->SecondPos) + 1)
                    res.push_back(Data::IndexedName::fromConst("Constraint", n));
            }
        }
        return res;

    auto getNames = [this, &silent, &res](const char *element) {  // <-- dead code starts here
        //...
```

A quick look makes me think it should be:
```cpp
std::vector<Data::IndexedName>
SketchObject::getHigherElements(const char *element, bool silent) const
{
    if (boost::istarts_with(element, "vertex")) {  // <-- removed the wrong indentation of the block
        std::vector<Data::IndexedName> res; // <-- moved ret inside the if block
        int n = 0;
        int index = atoi(element+6);
        for (auto cstr : Constraints.getValues()) {
            ++n;
            if (cstr->Type != Sketcher::Coincident)
                continue;
            if(cstr->First >= 0 && index == getSolvedSketch().getPointId(cstr->First, cstr->FirstPos) + 1)
                res.push_back(Data::IndexedName::fromConst("Constraint", n));
            if(cstr->Second >= 0 && index == getSolvedSketch().getPointId(cstr->Second, cstr->SecondPos) + 1)
                res.push_back(Data::IndexedName::fromConst("Constraint", n));
        }
        return res; // <-- moved return inside the if block
    }

    auto getNames = [this, &silent, &res](const char *element) {  // <-- not dead code anymore
        //...
```

Would be nice if someone with more knowledge could look at this.

### Full version info

```shell
latest main
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2025-04-11:
We should first check what it does. Where is the function being used?

hyarion, 2025-04-11:
Good question! It is overloading a function from GeoFeature but I don't see anywhere this is used in the codebase.

hyarion, 2025-04-13:
> Good spot! But rash change proposals are not helpful.

I was thinking aloud it's not a PR nor a proposal.

> In the code transferred to FC in a7bdbaa662ac8c79d900841770f2836918f09f64 there was a (commented out) enclosing if, which was removed in  b29abf7ffe22f9e7cb8e99a3c85afb4b1f89b975.
> ```
> //    if (testStatus(App::ObjEditing)) { // <=== deleted
>         if (boost::istarts_with(element, "vertex")) {
>             int n = 0;
>             int index = atoi(element+6);
>             for (auto cstr : Constraints.getValues()) {
>                 ++n;
>                 if (cstr->Type != Sketcher::Coincident)
>                     continue;
>                 for (int i=0; i<2; ++i) {
>                     int geoid = i ? cstr->Second : cstr->First;
>                     const Sketcher::PointPos &pos = i ? cstr->SecondPos : cstr->FirstPos;
>                     if(geoid >= 0 && index == getSolvedSketch().getPointId(geoid, pos) + 1)
>                         res.push_back(Data::IndexedName::fromConst("Constraint", n));
>                 };
>             }
>         }
>         return res;
> //    } // <=== deleted
> 
> ```
> Code was not reformatted and review (d92029f0bae4772ce90304b18ace5858f163136a) missed the early return

Thanks for digging in the history.

> (Kinda like Apple's goto fail).

Exactly what I was thinking when noticing it.

> Morals of the story: 
> The sooner uniform formatting project started by Werner is completed, the better. The early return would have been much more obvious.

Are you sure it would help and not just hide the issues more? Except for the return casing dead code, it is currently possible to see that indentation is wrong, by reformatting the code base that information would not be available.

And don't get me wrong, reformatting the code is important and should be done.

> But after all that, as you say, it appears the function is unused! Why is it so?

I've done a quick search and I haven't seen it being used anywhere.
But again this isn't a PR so I haven't done all the homework, I was hoping it could be a cooperative assignment.

hyarion, 2025-04-13:
When there's a return involve I agree, but when it's not, then it isn't as obvious that there is something strange going on.

FEA-eng, 2026-02-13:
@PaddleStroke Could you have a look at this?