FreeCAD/FreeCAD #20756: Part Design: "Select attachment" shows plane indicators for all components

furgo16, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Uncompress and open the [test file](https://github.com/user-attachments/files/19703579/Window.cg8.zip) with FreeCAD
2. Ensure the Part Design workbench is loaded and active
3. Double-click on the `Glass` object on the Tree view
4. Click on the "Create sketch" command on the toolbar
5. **Expected**: the "Select attachment" Task panel is shown, only the Glass body's planes are shown on the 3D view to pick
6. **Actual**: the "Select attachment" Task panel is shown, and... it's a Christmas tree.

[image]

The workaround to hide all the non-relevant plane indicators is to make the assembly in the model non-visible.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41225 (Git) Snap 1384
Build date: 2025/04/10 22:47:54
Build type: Release
Branch: main
Hash: 74ac46175d20ca945d4035768694f1536c93d941
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.11
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-11:
@PaddleStroke FYI

maxwxyz, 2025-04-11:
Can confirm, it is also not possible to select them.

PaddleStroke, 2025-04-14:
I don't know why you are complaining. It looks very nice ;)

PaddleStroke, 2025-04-14:
https://github.com/FreeCAD/FreeCAD/commits/main/src/Mod/PartDesign/Gui/TaskFeaturePick.cpp
So the temporary visibility is set in TaskFeaturePick::TaskFeaturePick line 161 of the file.

Now why this is happening.

I tested on 40500 and it's not happening. So it is quite recent indeed

closed 2025-04-15 by kadet1090