FreeCAD/FreeCAD #20757: Inconsistent naming of widgets in bottom right-hand corner area

furgo16, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The widgets in the bottom right-hand corner area of the UI (does it have a name?) have inconsistent denominations:
- Widget
- Indicator
- Area

[image]

I would suggest to unify the nomenclature and call them all "Indicator", but I'd love to hear more feedback.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41225 (Git) Snap 1384
Build date: 2025/04/10 22:47:54
Build type: Release
Branch: main
Hash: 74ac46175d20ca945d4035768694f1536c93d941
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.11
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-11:
That would be an option as well. And consistent with the context menu (RMB) of the toolbar area.

But even if not calling them with a noun in the context menu, it would be useful to agree on a denomination. At the very least for user documentation purposes and to avoid confusion with toolbars.

maxwxyz, 2025-04-13:
@FreeCAD/design-working-group any idea how this should be called?

semhustej, 2025-04-13:
I think it is a good practice to use the word "Widget", "Indicator", etc., because it makes communication easier:

- it educates users in a non-obtrusive way
- it is easier for user to search for the widget (or whatever the tool name) on wiki pages
- it is easier for users to file bugs or ask forum questions without ambiguity - "BIM status widget" is much clearer description than "BIM status in the lower right corner of the FreeCAD window"

pierreporte, 2025-04-13:
Widgets are differents from areas. I think that the notification area could be renamed notification zone. Navigation indicator is not really an indicator: it’s a selector with a tooltip. This one should be removed from the status bar completely and put in the settings. So we would end up with  just two names: widgets and one zone.

furgo16, 2025-04-13:
I agree that we want to bring uniformity and structure. For the sake of the discussion, we can consider there are two command ribbons:

- **Top**: mostly uniform, contains toolbar commands and the workbench selector. The widgets that can be shown/hidden are called toolbars
- **Bottom**: the area that is the subject of this issue, mixed content
 
That bottom ribbon, contains two types of widgets depending on their function:
- _Indicators_: provide visual or textual read-only information about a specific status or object (e.g. Notification area widget, Quick measure, zoom level, ...)
- _Quick access switches_: either through a command button or selector, they allow enabling a feature or status. Or switching between specific modes.

[image]

From left to right, the individual widgets are:

1. Textual information area about the currently selected or preselected object
2. A series of widgets that can be shown/hidden by the user:
   - Draft snap widget: quick access to most used snap funcions
   - BIM status widget: quick access to enable/disable a fairly mixed set of commands, some of them not BIM-specific
   - Draft scale widget: quick access to switch scale vale, provides also indicator function
   - Notification area: indicator, displays count and detail of user messages
   - Navigation indicator: quick access to switch navigation mode, provides also indcator function 
3. Quick measure: indicator, provides info about the selected entitiy' dimensions. Generally hidden unless there is a selection. When it appears, unfortunately it pushes most other widgets to the left, in an unexpected layout shift effect.
4. Combo zoom level indicator / units selector: both indicator (current zoom level) and quick access (document unit system selector). I could only thiink they were combined into one widget to save space in the bottom ribbon at some point, as they are not related to one another.

For consistency, I feel we should also consider:
- The user should be able to hide/show all widgets from the bottom ribbon, not just some of them.
- Zoom level and units selector should be decoupled into individual widgets. There is no relationship between the two that justifies the combination, which can be fairly confusing to users.

furgo16, 2025-04-14:
> You mean enable/disable the status area?

No, all the widgets from 1 to 4 in my previous message. Currently only those under point 2 can be enabled/disabled.

semhustej, 2025-04-15:
I ran into this issue https://github.com/FreeCAD/FreeCAD/issues/20785 with widgets and a somewhat related question from that issue arises:
Should workbench specific widgets be available to enable in different workbenches? In the case of OP, should there even be option to enable BIM and Draft widgets if for instance Part WB is active?

furgo16, 2025-04-16:
> why is it important to hide some?

To reduce visual clutter.

For instance, when the Draft workbench is active, and the Snap toolbar is active in the toolbar area (as it is by default), the Draft snap widget on the status bar shows a subset of the same widgets as the Snap toolbar. I also personally do not use the BIM status widget, so I generally hide it.

> I am also getting confused by possible overuse of "widget". In programming terms they are likely all widgets of one sort or another. From a user perspective the programming side is an irrelevant distraction. So what are they in userland words? Indicators is good. There may be others.

As a generic term to refer to all of these selectors and indicators GUI elements, widget is used here to give them a name for the sake of the discussion only. If I understand the remark correctly, I think we agree that "widget" should not be exposed to the user in the UI, as it's too generic and technical a term.