FreeCAD/FreeCAD #20758: BIM: formatting coordination task

marcuspollio, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Following the BIM, Arch and Native-IFC workbenches merge in 2024, a growing number of contributors are getting involved in the BIM module.
At an [early April 2025 dev meeting](https://github.com/FreeCAD/FreeCAD-developer-meetings/blob/4e051f94394b966c2cefd7c5a1cdede391f0c11d/Minutes/minutes-2025-04-06.md?plain=1#L58), it was agreed that having a consistent formatting for the module would be beneficial. The general idea is to adhere more closely to [PEP8 conventions](https://peps.python.org/pep-0008/) and prepare the module code so CI linters like [Black](https://github.com/psf/black) can be enabled.
The task here proposes a staged plan of action for such cleanups, refactorings and improvements.
PRs can be submitted by everyone. Just inform others when you start working on an item, thank you 😉

> [!NOTE]  
> The BIM module is here the main focus. A similar approach could be taken for the Draft module, as there are strong ties and similarities between both. Open for discussion how to proceed for Draft.

> [!IMPORTANT]  
> The list below is open for discussion and will be updated following detailed feedback.

 ### Formatting

To be tackled as much as possible during the current 1.1 dev cycle.
No functional or logic changes.

1. - [ ] License: make consistent blocks text and formatting. https://github.com/FreeCAD/FreeCAD/pull/20640
2. - [ ] License: add license blocks to files without
3. - [ ] Encoding: make sure all files are UTF-8 and remove declaration
4. - [ ] Indentation: make consistent
5. - [ ] Line length: make it fit `DEFAULT_LINE_LENGTH_LIMIT = 100` (see `tools/lint/black.py`)
6. - [ ] Line break: make consistent
7. - [ ] Blank lines: make consistent
8. - [ ] Module docstring: make position and content consistent
9. - [ ] Module dunder: make position and content consistent
10. - [x] Import: make one import per line. https://github.com/FreeCAD/FreeCAD/pull/20406
11. - [ ] Import: move imports to top when appropriate (not too heavy e.g. Part, ifc_tools, etc.)
12. - [ ] Import: make absolute import or explicit relative import, avoid wildcard imports
13. - [ ] String quotes: use double-quote, as per Draft and FEM coding conventions
14. - [ ] Whitespace: make consistent
15. - [ ] Trailing comma: make consistent
16. - [ ] Block comment: make consistent and add where needed
17. - [ ] Inline comment: make consistent and add where needed
18. - [ ] Guidelines: complete the module conventions and recommendations, based on the above experience

 ### Naming and larger refactoring

To be tackled probably during the next 1.2 dev cycle.
No functional or logic changes expected.

19. - [ ] Package and module name: use all-lowercase, avoid underscore if possible
20. - [ ] Class name: use CapitalizedWords (CamelCase)
21. - [ ] Function and method name: use lowercase_with_underscores()
22. - [ ] Variable name: use lowercase_with_underscores
23. - [ ] Constant name: use UPPERCASE_WITH_UNDERSCORES
24. - [ ] Non-public internal interface: prefix with a single leading underscore
25. - [ ] Arch vs BIM: decide the way forward and make consistent
26. - [ ] Guidelines: update the module conventions and recommendations, based on the above experience

### Full version info

```shell
do not apply
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marcuspollio, 2025-04-11:
Hi @maxwxyz 
Could you please make it a Task type and add the 1.1 milestone. TIA

maxwxyz, 2025-04-11:
I was too slow 😄

FEA-eng, 2026-03-02:
> * [ ]  Guidelines: complete the module conventions and recommendations, based on the above experience

So only this point is left?

Roy-043, 2026-09-25:
I think it would be good to move any remaining tasks to a new issue and close this one. For now I am changing the milestone to tbd.

marcuspollio, 2026-09-28:
Right, let's close this one. 1.1 is out since long.
The [remaining point](https://github.com/FreeCAD/FreeCAD/pull/27500) can be dealt on its own.

closed 2026-09-28 by marcuspollio