FreeCAD/FreeCAD #20761: BIM: blocking when building walls

gianlucafiore, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm working with LINUX MINT, I tried the weekly build version, and the 1.0 that comes in the program repository of the operating system.

When trying to use a very light base DXF file grouped into blocks, I trace over the walls. When I press Shift to force the path to an X or Y axis and follow a suggestion to snap to a vertex of the DXF, FreeCAD crashes completely, becoming unresponsive to anything in the interface, resulting in a forced shutdown.

I also noticed that it's starting to use up all the RAM.
[image]
 

[image]

[Proyecto.FCStd.zip](https://github.com/user-attachments/files/19711732/Proyecto.FCStd.zip)

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-04-11:
I am not able to reproduce on Windows. I tried with FC 1.0 and the latest weekly. Did you try in safe mode?

See behavior:

https://github.com/user-attachments/assets/ebea2ec5-5c15-4625-96b6-5be372335a98

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0

```

gianlucafiore, 2025-04-12:
Did you try doing several operations until the problem arises?

gianlucafiore, 2025-04-14:
It would be nice if someone could test it on Linux, I couldn't find the problem myself.

semhustej, 2025-04-15:
I tried in VM using Linux Debian 12. FreeCAD 1.1 build 41264.

I am not able to reproduce the issue.

semhustej, 2025-04-22:
This may be related. Seems similar, doesn't it?

https://github.com/FreeCAD/FreeCAD/issues/16485

closed 2025-12-15 by gianlucafiore