FreeCAD/FreeCAD #20766: Broken fillets (on circular geometry; symmetric part - one side works, sometimes)

Krakonos, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm building a part model and found a problem. On this model, I did two fillets. The part is symmetric. Fillet 1 works fine, but fillet 2 ends in a weird curve. I have seen this issue multiple times on other project, though they had much more complicated geometry. This is basically a cut down cylinder.

[image]

After some fiddling, Freecad crashed; but I don't have anything in coredumpctl to post and can't repro this crash. Might be unrelated, it's been open for a while.

I tried reproducing the same problem on the base geometry, but failed. Similar geometry, but the result is different for some reason - both fillets end up as garbage.

[image]

### Full version info

```shell
Tested on two builds - 19fc78e20 (from Arch Linux repo), but also the nightly appimage. Same results.

`FreeCAD_weekly-builds-41213-conda-Linux-x86_64-py311.AppImage`

OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.41213 (Git) Conda AppImage
Build date: 2025/04/09 15:48:15
Build type: Release
Branch: main
Hash: f258a2639cd8acd45cb101b7f34e90e603d91696
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/94.1104/1
Installed mods: 
  * freecad.gears 1.0.0
  * LCInterlocking
```

### Subproject(s) affected?

None

### Anything else?

Attaching both FCStd files.

[freecad-bug.zip](https://github.com/user-attachments/files/19714463/freecad-bug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-12:
Fillet bugs are quite common in FC because of an issue with the OpenCasCade kernel. Please make sure this hasn't been reported yet by x-reffing https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20label%3A%22Topic%3A%20Fillets%22

Krakonos, 2025-04-12:
I already looked through all the tickets in this search in the past year or two, didn't find anything that matches the first case - the interesting thing is the symmetrical nature of the object and one side working while the other doesn't. It almost looks like there is a sign issue somewhere and it then tries to rotate the surface resulting in those weird curves. No idea how the filleting engine works though.

The second case is pretty generic and may be a dupe.

closed 2025-04-13 by maxwxyz