FreeCAD/FreeCAD #20768: TechDraw: Annotation positions not saved when they have a View as Owner

psommerfeld, 2025-04-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the attached project:
1. Open "Page 1" in TechDraw
2. Move the various annotations. For instance see how they are distributed in the attached image
3. Save the project. Close it. Reopen it
4. Notice the changes to annotation positions were not saved.

It seems that annotations added in TechDraw that have a VIew as an owner do not keep their (x,y) position properly upon closing and opening the project.
In the attached pics, this first pic shows the annotations distributed in the views. The second pic shows what they look like after saving, closing and re-opening the project.
In the left view they seem to lose their proper y-position. In the right view, they lose their x-position.
If I remove the Owner from these annoations they seem to properly keep their position.

### Full version info

```shell
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.41138 (Git) Conda
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

### Subproject(s) affected?

TechDraw

### Anything else?

If the annotations all have just one view as owner, then it seem just x position is not saved.

[image]
[image]

[techdraw annos moving.FCStd.zip](https://github.com/user-attachments/files/19719373/techdraw.annos.moving.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

psommerfeld, 2025-04-12:
It seems the Owner does not specifically have to be a View object to not save properly.
And I'm not sure now that all objects that don't have an Owner save properly either.

Syres916, 2025-05-01:
@psommerfeld is it not the reason for the `Lock Position` boolean property that if the user wants any View or Annotation to stay in the exact position on the paper?

psommerfeld, 2025-05-01:
I don't understand how objects being rearranged between saving and reopening a project is anything but a bug.

> @psommerfeld is it not the reason for the `Lock Position` boolean property that if the user wants any View or Annotation to stay in the exact position on the paper?

WandererFan, 2025-05-01:
The first thing I notice is that all the Anno's attached to the BIM view are in the wrong place in the tree.   That is because the BIM view is a Symbol (svg) under the covers, and Symbols having children is not something that was allowed for.  Not a huge change to allow this.

I can not reproduce the Anno's losing their assigned position on save/restore with this AppImage:
```
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * fasteners 0.5.21
```

WandererFan, 2025-08-14:
@psommerfeld 
When I open the sample file and show the page, I see that all the annotations have a Y position of 
zero.

If I reposition the annotations on View or View1, save and reload, the annotations are where I left them.  

[image]