FreeCAD/FreeCAD #20769: techdraw "add cosmatic line" segfault

bhnasy15, 2025-04-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

in TechDraw bench
with a line and a vertix selected on a page view
clicking "add cosmatic parallel line" or "add cosmatic perpendicular line" will segfault.

### Full version info

```shell
OS: Arch Linux (wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

TechDraw

### Anything else?

Cosole output:
```console
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3dcd0) [0x7f9dabaf7cd0]
#1  0x7f9d3d71878e in execLineParallelPerpendicular(Gui::Command*, bool) from /usr/lib/freecad/lib/TechDrawGui.so+0x29e
#2  0x7f9daed3acbe in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x26e
#3  0x7f9daed3b1c7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x127
#4  /usr/lib/libQt6Core.so.6(+0x1b6cc9) [0x7f9dac116cc9]
#5  0x7f9dacaec63d in QActionGroup::_q_actionTriggered() from /usr/lib/libQt6Gui.so.6+0x5d
#6  /usr/lib/libQt6Core.so.6(+0x1b6cc9) [0x7f9dac116cc9]
#7  0x7f9dacaefb54 in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt6Gui.so.6+0x154
#8  /usr/lib/libQt6Widgets.so.6(+0x2e44d9) [0x7f9dad1554d9]
#9  /usr/lib/libQt6Widgets.so.6(+0x2e9533) [0x7f9dad15a533]
#10  0x7f9dacfc2b76 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xc66
#11  0x7f9dacf6fd9e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8e
#12  0x7f9dacf73ec6 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x10b6
#13  0x7f9daecca459 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#14  0x7f9dac0ba018 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
#15  0x7f9dacf6783c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21c
#16  /usr/lib/libQt6Widgets.so.6(+0x16cbc7) [0x7f9dacfddbc7]
#17  /usr/lib/libQt6Widgets.so.6(+0x16db90) [0x7f9dacfdeb90]
#18  0x7f9dacf6fd9e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8e
#19  0x7f9daecca459 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#20  0x7f9dac0ba018 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
#21  0x7f9dac6ff0c5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x3e5
#22  0x7f9dac7800c4 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x314
#23  0x7f9dac7802b7 in QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x197
#24  0x7f9dac1054aa in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x13a
#25  0x7f9dacf6fd9e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8e
#26  0x7f9daecca459 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#27  0x7f9dac0ba018 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
#28  0x7f9dac0ba3f2 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt6Core.so.6+0x382
#29  /usr/lib/libQt6Core.so.6(+0x3cfea8) [0x7f9dac32fea8]
#30  /usr/lib/libglib-2.0.so.0(+0x5e1e4) [0x7f9daa46e1e4]
#31  /usr/lib/libglib-2.0.so.0(+0xc1e97) [0x7f9daa4d1e97]
#32  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7f9daa46d615]
#33  0x7f9dac32d59d in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#34  0x7f9dac0c5376 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1f6
#35  /usr/lib/libQt6Widgets.so.6(+0x2eca22) [0x7f9dad15da22]
#36  /usr/lib/libQt6Widgets.so.6(+0x365b42) [0x7f9dad1d6b42]
#37  0x7f9dad1d746b in QToolButton::mousePressEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0x2eb
#38  0x7f9dacfc2756 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x846
#39  0x7f9dacf6fd9e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8e
#40  0x7f9dacf73ec6 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x10b6
#41  0x7f9daecca459 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#42  0x7f9dac0ba018 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
#43  0x7f9dacf6783c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21c
#44  /usr/lib/libQt6Widgets.so.6(+0x16c85f) [0x7f9dacfdd85f]
#45  /usr/lib/libQt6Widgets.so.6(+0x16db90) [0x7f9dacfdeb90]
#46  0x7f9dacf6fd9e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8e
#47  0x7f9daecca459 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#48  0x7f9dac0ba018 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
#49  0x7f9dac6ff0c5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x3e5
#50  0x7f9dac7800c4 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x314
#51  0x7f9dac7802b7 in QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x197
#52  0x7f9dac1054aa in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x13a
#53  0x7f9dacf6fd9e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8e
#54  0x7f9daecca459 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#55  0x7f9dac0ba018 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
#56  0x7f9dac0ba3f2 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt6Core.so.6+0x382
#57  /usr/lib/libQt6Core.so.6(+0x3cfea8) [0x7f9dac32fea8]
#58  /usr/lib/libglib-2.0.so.0(+0x5e1e4) [0x7f9daa46e1e4]
#59  /usr/lib/libglib-2.0.so.0(+0xc1e97) [0x7f9daa4d1e97]
#60  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7f9daa46d615]
#61  0x7f9dac32d59d in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#62  0x7f9dac0c5376 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1f6
#63  0x7f9dac0bd159 in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0xb9
#64  0x7f9daebd15d4 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xbd4
#65  freecad(+0x7e74) [0x55b4a6a13e74]
#66  /usr/lib/libc.so.6(+0x27488) [0x7f9dabae1488]
#67  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f9dabae154c]
#68  freecad(+0x8275) [0x55b4a6a14275]

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FartherOut, 2025-04-12:
Tried it in Windows 11 with the x86_64 portable version and both cosmetic lines seems to work properly.

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Norwegian Bokmal/Norway (nb_NO)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

ErtugrulEge, 2025-04-12:
Maybe Linux related

luzpaz, 2025-04-12:
@bhnasy15 can you test on dev[^1] build and see if you can reproduce ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

xtemp09, 2025-04-13:
I cannot reproduce, or, at least, don't know how.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41236 (Git)
Build date: 2025/04/12 03:48:47
Build type: Debug
Branch: master
Hash: e990b456b9db3952cf785cb38619622415b13dca
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

bhnasy15, 2025-04-13:
I Tried the weekly build, same thing.
I opend the same file, in the TechDraw bench the "add cosmatic parallel line" function seg faults.

```console
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
```

console output:
```console
$ ./FreeCAD_weekly-builds-41232-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR41232 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3dcd0) [0x7f3150ac1cd0]
#1  0x7f30b46a75d4 in execLineParallelPerpendicular(Gui::Command*, bool) from /tmp/.mount_FreeCAaLmKLc/usr/lib/TechDrawGui.so+0x234
#2  0x7f31539a699f in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libFreeCADGui.so+0x21f
#3  0x7f31539a6d4f in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libFreeCADGui.so+0x12f
#4  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6(+0x30a527) [0x7f31511c0527]
#5  0x7f3151a3dd20 in QActionGroup::triggered(QAction*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Gui.so.6+0x40
#6  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6(+0x30a527) [0x7f31511c0527]
#7  0x7f3151a39583 in QAction::triggered(bool) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Gui.so.6+0x43
#8  0x7f3151a3c972 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Gui.so.6+0xf2
#9  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6(+0x3c2275) [0x7f3152104275]
#10  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6(+0x43c452) [0x7f315217e452]
#11  0x7f31522f5d94 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0xa64
#12  0x7f31522e60f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0x84
#13  0x7f31522e6f8d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0xddd
#14  0x7f3153949009 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libFreeCADGui.so+0xc9
#15  0x7f315120329a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6+0x16a
#16  0x7f31522ee398 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0x1e8
#17  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6(+0x5b232e) [0x7f31522f432e]
#18  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6(+0x5b5930) [0x7f31522f7930]
#19  0x7f31522e60f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0x84
#20  0x7f3153949009 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libFreeCADGui.so+0xc9
#21  0x7f315120329a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6+0x16a
#22  0x7f315166fcc0 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Gui.so.6+0x810
#23  0x7f31516cd324 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Gui.so.6+0xb4
#24  /tmp/.mount_FreeCAaLmKLc/usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x5874c) [0x7f314b69074c]
#25  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/./libglib-2.0.so.0(+0x5b5e6) [0x7f314f43d5e6]
#26  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/./libglib-2.0.so.0(+0x5e7b7) [0x7f314f4407b7]
#27  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f314f440ece]
#28  0x7f3150fa5d0e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6+0x6e
#29  0x7f315123b7fc in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6+0x1ac
#30  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6(+0x439cf3) [0x7f315217bcf3]
#31  /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6(+0x2ee6b2) [0x7f31520306b2]
#32  0x7f3152030c8c in QToolButton::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0x1cc
#33  0x7f31522f5dbc in QWidget::event(QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0xa8c
#34  0x7f31522e60f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0x84
#35  0x7f31522e6f8d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Widgets.so.6+0xddd
[... cut ...]
#52  0x7f315123bc0a in QCoreApplication::exec() from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libQt6Core.so.6+0x9a
#53  0x7f31538a4124 in Gui::Application::runApplication() from /tmp/.mount_FreeCAaLmKLc/usr/bin/../lib/libFreeCADGui.so+0x734
#54  /tmp/.mount_FreeCAaLmKLc/usr/bin/freecad(+0x5ca0) [0x5562226ecca0]
#55  /usr/lib/libc.so.6(+0x27488) [0x7f3150aab488]
#56  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f3150aab54c]
#57  /tmp/.mount_FreeCAaLmKLc/usr/bin/freecad(+0x602a) [0x5562226ed02a]
```

Syres916, 2025-04-18:
I upgraded EndeavorOS from:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41308 (Git)
Build date: 2025/04/18 04:41:18
Build type: Release
Branch: main
Hash: 6be09c29f6217c6338cf09dcc9a658454287bd83
Python 3.13.2, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: User_dark.qss/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.075/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

to:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41308 (Git)
Build date: 2025/04/18 04:41:18
Build type: Release
Branch: main
Hash: 6be09c29f6217c6338cf09dcc9a658454287bd83
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: User_dark.qss/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.075/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

in order to get as similar dependencies as the OP and neither build segfaulted so I would tend to want to eliminate Wayland first.

bhnasy15, 2025-04-26:
I tried it on X11 with i3, still did not work
```
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```
```
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Template File: /tmp/.mount_cadkbEaIg/usr/share/Mod/TechDraw/Templates/A4_Landscape_blank.svg is not readable
/usr/include/c++/14.2.1/bits/stl_vector.h:1130: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Base::Vector3<double>; _Alloc = std::allocator<Base::Vector3<double> >; reference = Base::Vector3<double>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```
However, the latest conda version works just fine on wayland and X11:
```
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```
I am happy with the conda build, it works great.
feel free to close the this report, sorry if I wasted your time.

closed 2025-04-26 by bhnasy15