FreeCAD/FreeCAD #20771: Sketcher: Constraint coincident tool gives error when applying two points on the same  line

ErtugrulEge, 2025-04-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, Constraint coincident tool gives error when applying two points on the same  line 

1. Draw a simple rectangle  or line
2. Select a point 
3. Select other point on the same line
4. Apply Constraint coincident tool

[image]

also after error message you cant edit the sketch you have to delete the sketch

[image]

```
18:47:37  Updating geometry: Error build geometry(0): Both points are equal
18:47:37  Invalid solution from DogLeg solver.
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-08-24:
The error is also triggered for other constrains. If you make opposite sides of the rectangle tangent, you get the same error, but it sort of works if you shake the rectangle.

PaddleStroke, 2026-09-07:
This bug is fixed by https://github.com/FreeCAD/FreeCAD/pull/25304

PaddleStroke, 2026-09-07:
Erratum this can still happen when you select points not of the same line

closed 2026-09-08 by kadet1090