FreeCAD/FreeCAD #20772: Pocket extrude of simple rectangle fails in complex body

w5xd, 2025-04-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The attached zip has the file WWVBclockEnclosure2.FCStd. The last feature, Pocket011 in the one-and-only body in that file is a Pocket extrude of a simple rectangle in Sketch024. The result is definitely not rectangular cut in the body: it is a shape not easily described.  I tried and failed to create a simplified model that reproduces this issue--I failed to guess what of the couple of dozen features leading up to this pocket confused freecad. So I am adding the whole project to the bug report. 

[pocketextwrong.zip](https://github.com/user-attachments/files/19720735/pocketextwrong.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * OpticsWorkbench 1.1.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-15:
Could you add a screenshots and maybe a simpler file? Please also test in the weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

leoheck, 2025-04-16:
When the `trim LCD front panel` it produces a lot of problems.
I think the design is faulty, than the last pocket because you don't have a valid model anymore.
Maybe if you fix that part, and any possible other parts, things may work better. 

[image]

leoheck, 2025-04-16:
This design has so many problems that it is easier to restart it from scratch.

luzpaz, 2025-10-26:
Thank you but we also need your full [About](https://wiki.freecad.org/About) info, thanks in advance

w5xd, 2025-10-26:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.299/1
Installed mods: 
  * OpticsWorkbench 1.1.2

leoheck, 2025-10-26:
@schmidtw check whether your design is broken with `Analyze Geometry`. Boolean operations require good geometry, and sometimes a feature messes up with the geometry even if we they the result is fine. The initial example in this post is quite a mess. Everything is broken, so it looks kind of obvious that the feature will fail.

FEA-eng, 2026-10-21:
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

maxwxyz, 2026-10-21:
So it looks correct I assume

closed 2026-10-21 by maxwxyz

luzpaz, 2026-10-21:
@w5xd can you confirm fix using latest development build ?