FreeCAD/FreeCAD #20773: [Documentation / FAQ] Is .dwg supported by FreeCAD?

rubyFeedback, 2025-04-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am trying to open a .dwg file; someone else sent it to me via an email. He is using
windows, I am using Linux.

I read on StackOverflow that FreeCAD may be able to open the .dwg file.

Anyway. It would be nice if you guys could:

a) add a FAQ, and link to it from the main README (could also be on the 
wiki here; but the word "FAQ" should ideally also appear in the main README
so it is easier for us to find it)

and then

b) also document there whether .dwg is supported or not. This would help,
in that we know whether FreeCAD would be able to view it or not.

### Full version info

```shell
-
```

### Subproject(s) affected?

None

### Anything else?

-

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-04-13:
The b) part is answered on FreeCAD wiki: https://wiki.freecad.org/FreeCAD_and_DWG_Import

luzpaz, 2025-04-13:
@Reqrefusion quick question, should we keep this open?

Reqrefusion, 2025-04-13:
> [@Reqrefusion](https://github.com/Reqrefusion) quick question, should we keep this open?

[image]

It is stated everywhere that there is something related to the formats in which the DWG format is supported.  It's also on FreeCAD's wikipedia page. I don't know where there could be more.

closed 2025-04-14 by Roy-043