FreeCAD/FreeCAD #20776: Mirroring a component with Scale = -1 doesn't work anymore

MiqCG, 2025-04-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Mirroring a component with Scale = -1 (Link feature) was working until some weeks ago. I used it in the integrated assembly WB [1] and outside it, but now (I don't know exactly when) the behavior has changed and the mirroring fails after transforming or dragging the part; then the component turns black (similar to this [issue](https://github.com/FreeCAD/FreeCAD/issues/20464) ).

Steps to reproduce the issue:
1.     open the Original_Arm file (attached)
2.     open a new file and save the new file
3.     link the object to the new file with `Make link`
4.     apply `scale = -1` in the properties panel (the mirrored part appears)
5.     transforming (moving) the object make the component return to the original state and the object turns black.

It also affects the insert tool in the integrated assembly WB. See screencast below. 
It works properly with appimage 39708 (see the second screencast) but it didn't work work yet with appimage 41127 (~March 2025). This issue has already been confirmed in https://github.com/FreeCAD/FreeCAD/issues/20464#issuecomment-2799566825

The impossibility to obtain a mirrored component with another procedure is described [here](https://github.com/FreeCAD/FreeCAD/issues/14117) and [here](https://github.com/FreeCAD/FreeCAD/issues/5705) and they are still unsolved.

Example files:
[issue 20464 Mirroring.zip](https://github.com/user-attachments/files/19718615/issue.20464.Mirroring.zip)

Not working with appimage 41213:
https://github.com/user-attachments/assets/43450e25-1475-46a5-9a89-46853af8a296

Working in appimage 39708:
https://github.com/user-attachments/assets/e6d73a1e-3627-4218-a612-c57c70a9ea16

[1] https://github.com/FreeCAD/FreeCAD/issues/14623
[2] https://forum.freecad.org/viewtopic.php?style=3&p=555675#p555675

### Full version info

```shell
`OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb) Architecture: x86_64 Version: 1.1.0dev.41213 (Git) Conda AppImage Build date: 2025/04/09 15:48:15 Build type: Release Branch: main Hash: f258a2639cd8acd45cb101b7f34e90e603d91696 Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1 Locale: Catalan/Spain (ca_ES) Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/ Logical DPI/Physical DPI/Pixel Ratio: 96/93.8695/1.07292`
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MiqCG, 2025-04-14:
I have been testing the example files in more [appimages](https://github.com/FreeCAD/FreeCAD/issues/20464#issuecomment-2800063757), thanks to @leoheck, and the problem was already existing in appimage 40408.

Could it be related to the modifications of the `Transform` tool that weren't present in revision 39708?

[image]

maxwxyz, 2025-04-15:
@FreeCAD/cad-working-group should this be possible, -1 scale to mirror an object is not a good concept.

kaiwas, 2025-04-15:
> is not a good concept

This may be bad, but it is the only way to mirror an object without leaving its original copy.
By the way, I was never able to do this using macros. Unfortunately.

https://github.com/FreeCAD/FreeCAD/issues/16581

jffmichi, 2025-04-15:
I don't really get why this would be "not a good concept" or should be avoided. Every mirror operation in FreeCAD is doing exaclty that, just with a different name. It's perfectly well defined and also used in other applications, e.g. Blender:

[image]

kaiwas, 2025-04-15:
@jffmichi Changes that correspond to the methods of "direct" modeling are accepted with difficulty. (You can look at the comments in my issue. The link is above and it will become a little clearer)

jffmichi, 2025-04-15:
For now I'm only arguing to keep the whole range of values valid for properties that already exists. I think adding scale values to the placement or something like that is a broader discussion (would probably not be against it)...

MiqCG, 2025-05-29:
> Where are we at on this ?

I guess that the problem is still there, the lack of alternative methods is still there but I don't know where is the discussion.
Related to the discussion, my question is if the feature was removed intentionally or by accident. If the first is the case, someone will know why and the implications of not removing it, but if it was the second, why not to recover it meanwhile the discussion brings us somewhere?

leoheck, 2025-05-30:
I could scale. It initially works.
But the operation is instable. If something changes.... like for instance moving the object, it reverts the operation.

[image]

Here it was transformed downwards... then the mirror effect was reverted, and it became black.

[image]

```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

leoheck, 2025-05-30:
There are 2 problems.
- The operation is unstable; it tends to be undone.
- The color is damaged

MiqCG, 2025-05-30:
> * The operation is unstable; it tends to be undone.
> 
> * The color is damaged

Yes, this is the initial behaviour I reported.

luzpaz, 2025-06-18:
Do we know what the regression is from ?

leoheck, 2025-06-19:
It appears to be related to links. I guess. But who knows?
@maxwxyz, if you follow this report to the end, it may point it to be related to the Transform tool as @MiqCG said.

Considering this.. 
> Working in appimage 39708
> Not working with appimage 41213

```
~/Applications took 3.4s …
➜ ./freecad_version_to_git_hash 39708
003e239daf

~/Applications took 3.6s …
➜ ./freecad_version_to_git_hash 41213
4c7fb9625b
```

Here quick steps to reproduce the issue.
1. Open a file with a Body, can be the [Original_Arm](https://github.com/user-attachments/files/19718615/issue.20464.Mirroring.zip) from the post.
2. Click on this Original_Arm.
3. Then click make a link of it.
4. Scale the link to -1
5. Transform the link to see the scaling operation breaking.

I have some extra Appimages, so I cpuld narrow down the search a bit

- freecad_weekly_40408
- freecad_weekly_40444
- freecad_weekly_40655
- freecad_weekly_40971
- freecad_weekly_40971
- freecad_weekly_41127
- freecad_weekly_41138
- freecad_weekly_41213

The first one (40408) already gave me the same broken results as the 41213.
If someone has another version closer to 39708, it could help a lot.

```
~/Applications took 3.6s …
➜ ./freecad_version_to_git_hash 41213
49c2888cde
```

Perhaps we can using this to find something meaningfull.

Here are some atempts find something:

```
➜ git log 003e239daf..49c2888cde --oneline | grep -i Transform
d5b0f686e1 [Part TopoShapeExpansion] move shape transform from make element copy into ruled surface function since it does not work well when copying vertex objects
1e573241c5 FEM: Constraint transform - selection error message update (#19192)
4ba83f0f50 [TD]use transformShape instead of transformGeometry
7bb4302eaf Merge pull request #17564 from kadet1090/new-transform-dialog
178535980e Gui: Use extrinsic angles for absolute position mode in transform dialog
27c1eaa39c Gui: Disable controls when selecting origin in transform dialog
fe96e7e194 Gui: Reset UVW coordinate system in transform when rotation changes
a3bf63bae2 PD: Forward transform requests to Body
25cf119c60 Gui: Add Dragger placement preview in Transform dialog
1df3b5be6c Gui: Rework TaskCSysDragger into new Transform Dialog

➜ git log 003e239daf..49c2888cde --oneline | grep -i scale    
dd3ae90fe7 Draft: task_scale.py fix circular imports (#19574)
3202dde7ca PD(hole taskpanel): Add custom widget for font-scaled SVGs
b95b500e5b TechDraw: fix editable/autofill scale in Templates
d587e6a91a Draft: make move, rotate and scale commands link-aware (#18795)

➜ git log 003e239daf..49c2888cde --oneline | grep -i scaling
5bcc0d9612 Merge pull request #19477 from Roy-043/Draft-Introduce-Edit-option-for-Draft-Clone-scaling
c88b44f0da Draft: Introduce Edit option for Draft Clone scaling
[... 43 lines cut ...]

Perhaps we have 2 issues overlapping here.
1. Color breaks. The Object turns to Black-ish.
2. Reverse operation is reset, after Transformation. It restores the broken invert and colors after the file is reopened.

Using Asm4. I can mirror a part. And it starts black-ish, from the start.
Ah, now it has new problem. The original and the mirrored version are being displayed overlapping each other.
However I can move that part. 

Here is an example

- 1st object is the original one, moved to the left of the origin a bit.
- 2nd object is the link of the 1st one, scaled to -1, placed were it was created.
- 3rd object is the link of the 1st one, scaled to -1, translated rightward.
- 4rth object was mirrored using Asm4. It is black-ish by the start, now it has this new problem where the original overlaps it.

[image]

- 3rd object, after reopening the file. The operation is restored. This seems to indecate one of the issues is in the Transform tool.

[image]

Here is the full picuture, including Object tree, and my modified sample file [Original_Arm.FCStd.zip](https://github.com/user-attachments/files/20822563/Original_Arm.FCStd.zip)

[image]

leoheck, 2025-06-20:
It is not working on this version here too `weekly-2025.04.21~500`

leoheck, 2025-06-21:
Tried to find where this issue began... I think it started to appear on this commit `weekly-2025.04.21~843`.

```bash
# weekly-2025.04.21~484 | 49c2888cde                               | Not working | 1.1.0devR40408 <==== My oldest AppImage
# weekly-2025.04.21~500 | e80ee07d63fc0bc378b932885dc517f44999fee3 | Not working | 1.1.0devR
# weekly-2025.04.21~525 | a6c92e19f6bc3e13b8da3332944b41c011bbb33c | Not working | 1.1.0devR40397
# weekly-2025.04.21~537 | 2780c9b13e3a4980c92beb71dd4c2e443e455941 | Not working | 1.1.0devR40380
# weekly-2025.04.21~543 | 2c485cf99821c9220418cbff5370ad223f96a707 | Not working | 1.1.0devR40354 
# weekly-2025.04.21~546 |                                          | Not working | 1.1.0devR40346
# weekly-2025.04.21~547 |                                          | Not working | 1.1.0devR40345
# weekly-2025.04.21~548 |                                          | Not working | 1.1.0devR40344
# weekly-2025.04.21~549 |                                          | Not working | 1.1.0devR40343
# weekly-2025.04.21~550 | a2827acf7e39f128a06c6883348b36361c5ce2f2 | Not working | 1.1.0devR40342
# weekly-2025.04.21~600 |                                          | Not working | 1.1.0devR40272
# weekly-2025.04.21~633 |                                          | Not working | 1.1.0devR40219
# weekly-2025.04.21~649 |                                          | Not working | 1.1.0devR40143
# weekly-2025.04.21~657 |                                          | Not working | 1.1.0devR40127
# weekly-2025.04.21~661 |                                          | Not working | 1.1.0devR40122
# weekly-2025.04.21~663 |                                          | Not working | 1.1.0devR40115
# weekly-2025.04.21~664 |                                          | Not working | 1.1.0devR40113
# weekly-2025.04.21~665 |                                          | Not working | 1.1.0devR40110
# weekly-2025.04.21~666 | 235f8de27f791c240b40cf85368e2b1b0f54bf9b | Not working | 1.1.0devR40109
# weekly-2025.04.21~799 |                                          | Not working | 1.1.0devR39936
# weekly-2025.04.21~832 |                                          | Not working | 1.1.0devR39849
# weekly-2025.04.21~840 |                                          | Not working | 1.1.0devR39841
# weekly-2025.04.21~842 |                                          | Not working | 1.1.0devR39839
# weekly-2025.04.21~843 | 7bb4302eaf                               | Not working | 1.1.0devR39838
# weekly-2025.04.21~844 |                                          | Working     | 1.1.0devR39818
# weekly-2025.04.21~848 |                                          | Working     | 1.1.0devR39814
# weekly-2025.04.21~865 |                                          | Working     | 1.1.0devR39788
# weekly-2025.04.21~932 | 003e239daf                               | Working     | 1.1.0devR39708 ? <=== Oldest working AppImage
```

leoheck, 2025-06-21:
This commit https://github.com/FreeCAD/FreeCAD/commit/7bb4302eaf5cf2c7a70b9b318bd204bbcfd8b477 is big, so maybe someone could look at it, right @maxwxyz ?

@chennes could you give it a quick check to point where this regression might be happening?

leoheck, 2025-06-21:
I am narrowing this search a bit further...

```
# weekly-2025.04.21~843^2    | not working **
# weekly-2025.04.21~843^2~9  | not working
# weekly-2025.04.21~843^2~11 | building...
# weekly-2025.04.21~843^2~13 | working
# weekly-2025.04.21~843^2~17 | working
# weekly-2025.04.21~843^2~18 | working
```

leoheck, 2025-06-21:
```markdown
commit fab235682db031e99bf7d08ca7eaee09aeb1fb04 (HEAD)
Author: Kacper Donat <kadet1090@gmail.com>
Date:   Thu Dec 5 14:42:13 2024 +0100

    Gui: Make ViewProviderLink based on ViewProviderDragger

src/Gui/TaskCSysDragger.cpp
src/Gui/ViewProviderDragger.cpp
src/Gui/ViewProviderDragger.h
src/Gui/ViewProviderLink.cpp
src/Gui/ViewProviderLink.h
src/Gui/ViewProviderLinkPy.xml
src/Gui/ViewProviderLinkPyImp.cpp
src/Mod/Part/App/Attacher.cpp
src/Mod/Part/App/Services.cpp

commit 1df3b5be6c6343d0fd2cb1ee3e8410c5e029d0de
Author: Kacper Donat <kadet1090@gmail.com>
Date:   Sun Oct 27 18:39:51 2024 +0100

    Gui: Rework TaskCSysDragger into new Transform Dialog
    
    This commit refactor ViewProviderDragger and TaskCSysDragger to be more
    modern and to support selecting TransformOrigin.

src/App/Application.cpp
src/App/CMakeLists.txt
src/App/GeoFeature.cpp
src/App/GeoFeature.h
src/App/Services.cpp
src/App/Services.h
src/Base/Rotation.cpp
src/Base/Rotation.h
src/Gui/CMakeLists.txt
src/Gui/SoFCCSysDragger.cpp
src/Gui/SoFCCSysDragger.h
src/Gui/SoTextLabel.cpp
src/Gui/SoTextLabel.h
src/Gui/TaskCSysDragger.cpp
src/Gui/TaskCSysDragger.h
src/Gui/TaskCSysDragger.ui
src/Gui/Utilities.h
src/Gui/ViewProvider.cpp
src/Gui/ViewProviderDragger.cpp
src/Gui/ViewProviderDragger.h
src/Gui/ViewProviderLink.cpp
src/Mod/Part/App/AppPart.cpp
src/Mod/Part/App/CMakeLists.txt
src/Mod/Part/App/Services.cpp
src/Mod/Part/App/Services.h
```

kadet1090, 2025-06-21:
@leoheck thanks for bisecting! I'll check out why it does not work.

leoheck, 2025-06-22:
Sure, building it here. @kadet1090

leoheck, 2025-06-22:
I confirm that it fixes the Link Transform works, after Link Scale<0. 
Thank you @kadet1090 

However, there is still something missing. It does not fix the broken color when using Asm4 > Mirror, which uses Links_Sacaling. 

I will have to check this deeper later today.

This image shows Asm4, a mirror operation of a mirror operation.
The 1st mirror produces a Black object, and the second Mirror (of the first mirror). Fixes the 3rd and breaks the 4th object.

[image]

closed 2025-06-23 by chennes

furgo16, 2025-06-23:
> It looks like this color issue with the Asm4 mirror predates weekly-2025.04.21~932 and has nothing to do with this new Transform widget. 

I can confirm that. Back when I was using Assembly 4, the black mirrored objects were a known issue. I seem to remember there was a workaround for it as well. A quick forum search returns: https://forum.freecad.org/viewtopic.php?t=84570

leoheck, 2025-06-23:
@furgo16 the workround works with 0.21, but after that, since the settings page changed, I could not see the workaround being fixed anyone.

leoheck, 2025-06-24:
@furgo16 I think i found the issue why this workaround does not work in 1.1

Here is an example of a mirror of a mirror. The visibility is already set to `Two side`.
But there is an Enum nested with 2 items in it, starting with `One side`, and then `Two side`
I am not sure if this is right. But then the fix does not work.

[image]

However, if I remove the `One side` from the enum, or if I move it to the end of the list, it fixes the issue, see.

[image]

Do you know what this enum does? Is it right to have it nested there?

PR #22127 (2025-06-23): Gui: Use proper placement property for Link
Links require different placement property (`LinkPlacement`) to be used, otherwise it breaks the transform.

## Issues
Fixes: #20776
commit d35b6380 (2025-06-23): Gui: Use proper placement property for Link

Links require different placement property (LinkPlacement) to be used,
otherwise it breaks the transform.

Fixes: #20776