FreeCAD/FreeCAD #20779: Sketcher behaving abnormally

YosypC, 2025-04-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sorry for the lack of a proper title, I really don't know how to name this.

[image]

[image]

MainFrame -> Pocket007

Trying to change the radius of the upper right lip of the left shape yields the errors in the screenshot. The lines remain white, and it appears I can change said radius manually after deleting the constraint. Trying to radius constrain it results in it magically disappearing after confirming a length.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Curves 0.6.61
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

[RenameToFCStdExtension.zip](https://github.com/user-attachments/files/19726171/RenameToFCStdExtension.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-04-13:
Radius can not be less than 3 mm

[image]

FEA-eng, 2025-10-07:
@tarman3 Should this issue be closed?

luzpaz, 2025-12-09:
cc @tarman3 

Also, can someone please propose a less generic title for this ticket ? Thank you!