FreeCAD/FreeCAD #20782: FreeCAD Wiki slow / not reachable

semhustej, 2025-04-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

For the past 24 hours, FreeCAD Wiki has been very slow (loading time of a page above 1 minute) or not reachable.

[image]

Currently it is a hit and miss
`Apr 14 08:36:56 2025 UTC`

I tested from two devices with two different internet providers with the same result.

### Full version info

```shell
-
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Maybe it's the same AI problem as with forum: https://github.com/FreeCAD/FreeCAD/issues/19062

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-14:
I can confirm

semhustej, 2025-04-14:
It is working OK now.

`Apr 14 15:35:45 2025 UTC`

closed 2025-04-15 by maxwxyz

thelowkeyguy, 2025-04-15:
I am also getting 502 Bad Gateway. When it open is extremely slow  (São Paulo - BR - IP address).  Is there any other place where I can download PDFs versions of documents hosted on FreeCAD wiki ?

FlachyJoe, 2025-04-15:
@thelowkeyguy Web.archive.org stores a snapshot from a few days https://web.archive.org/web/20250409025011/https://wiki.freecad.org/

semhustej, 2025-04-15:
Works fine for me (Europe) now.

`Apr 15 14:38:11 2025 UTC`

semhustej, 2025-04-18:
Wiki slow or not reachable right now again.

`Apr 18 14:15:46 2025 UTC`

semhustej, 2025-04-18:
Works fine now.

`Apr 18 16:37:28 2025 UTC`

semhustej, 2025-04-23:
Wiki still not reachable 18 hours later.

`Apr 23 08:05:40 2025 UTC`

Does it make sense to report these problems here?

semhustej, 2025-04-25:
Wiki currently not reachable again:

`Apr 25 15:42:01 2025 UTC`