FreeCAD/FreeCAD #20783: Assembly: unable to save file when a joint is frozen

MiqCG, 2025-04-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the integrated assembly WB, you can freeze joints to temporarily disable its influence. But , when trying to save the file with frozen joints, this error appears:

```
11:48:31  Failed to auto-save document 'xxx_file_name_xxx'
11:49:01  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADAbort'>: At least one object is frozen, unable to save.
```

If working with a file with several frozen joints, this limitations is inconvenient.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41213 (Git) Conda AppImage
Build date: 2025/04/09 15:48:15
Build type: Release
Branch: main
Hash: f258a2639cd8acd45cb101b7f34e90e603d91696
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.8695/1.07292
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-14:
Assembly maintainers should consider implement the Suppression capability.

The Frozen state is intended to block a recalculation so that the object cannot be modified. If you freeze a joint, its property changes are ignored (e.g. you change an expression of the distance but it keeps the old value), but it should still work.

PaddleStroke, 2025-07-09:
I don't know well the "Freeze" feature. But from what I see this is not an assembly issue. You cannot save a file if any object is "Freeze". It has nothing to do with assembly or joints.

Having said that, right clicking on a joint show "Freeze" as first option.

[image]

Which is kind of misleading since it's not even clear what "Freeze" is. So users are likely to think that this is a way to deactivate a joint.

closed 2025-07-21 by maxwxyz