FreeCAD/FreeCAD #20784: CenterOfMass of Tube not available

ShuffleWire, 2025-04-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm using Cylinder and Tube to model object, and found out during Inertia estimation that the Tube don't have CenterOfMass nor MatrixOfInertia, contrary to the Cylinder (or the Cube, for instance)
I'm working with basic Part Box/Cylinder/Tube, and query those using the Python API

Using the linked file (But that easy the reproduce : empty file, and Using Part workbench, add a Cube, Cylinder and Tube)

Python commands : 
```
>>> doc = App.ActiveDocument
>>> s=doc.getObject("Box").Shape
>>> s.CenterOfGravity
Vector (5.0, 5.0, 5.0)
>>> s.CenterOfMass
Vector (5.0, 5.0, 5.0)
>>> s.MatrixOfInertia
Matrix ((16666.7,0,0,0),(0,16666.7,0,0),(0,0,16666.7,0),(0,0,0,1))
>>> s=doc.getObject("Cylinder").Shape
>>> s.CenterOfGravity
Vector (-5.937035405398566e-16, -5.7, 5.000000000000001)
>>> s.CenterOfMass
Vector (-5.937035405398566e-16, -5.7, 5.000000000000001)
>>> s.MatrixOfInertia
Matrix ((1172.86,8.62993e-14,-1.77636e-14,0),(8.62993e-14,1172.86,1.93908e-13,0),(-1.77636e-14,1.93908e-13,251.327,0),(0,0,0,1))
>>> s=doc.getObject("Tube").Shape
>>> s.CenterOfGravity
Vector (-10.500000000000002, 1.2859059199447271e-16, 5.0)
>>> s.CenterOfMass
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: 'Part.Compound' object has no attribute 'CenterOfMass'
>>> s.MatrixOfInertia
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: 'Part.Compound' object has no attribute 'MatrixOfInertia'
```

I would expect that those computations would work the same way, been rather easy.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly3 0.12.2
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-14:
For a Compound, each solid has a MatrixOfInertia.
Tube is a one-solid compound so `App.ActiveDocument.Tube.Shape.Solids[0].MatrixOfInertia` give the result.

ShuffleWire, 2025-04-14:
That is correct, and as well for others solid, there is a `Solids` member. But I would expect that, as for others solids, `CenterOfMass` and `MatrixOfInertia` to be defined