FreeCAD/FreeCAD #20785: Draft Widgets not preserved in BIM workbench after workbenches are switched

semhustej, 2025-04-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Draft Widgets are not preserved in BIM workbench after workbenches are switched.

Steps to reproduce:

- open new file with the BIM workbench selected
- turn `Draft Snap Widget` and `Draft Scale Widget` on
- both widgets are turned on
- switch to Draft workbench
- both widgets are still turned on
- switch to BIM workbench
- both widgets are turned off and user has to enable them again

See behavior in the video:

https://github.com/user-attachments/assets/4c89f1ac-ee2c-422c-8dbf-7b974d9bc9f6

The same happens with `BIM Status Widget` when it is enabled in Draft WB and than workbenches are switched back and forth.

I think it is quite reasonable to expect users wanting to use Draft WB widgets. I am not sure if Draft users would want to use the BIM widget, but it does not hurt to have the option.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

Draft

### Anything else?

I was able to reproduce the behavior in FreeCAD 1.0

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-04-14:
@semhustej I got repro. Interestingly enough, if you run BIM Mod first, the options aren't even visible (at least for me). You have to initialize Draft GUI at least once to make those checkboxes visible in BIM. Same if you run Draft first, `BIM Status` checkbox is not even visible for me. Is it the same for you?

Enabling those options for both sides is pretty easy, I can make a PR but I wonder if having those checkboxes visible at all is not a bug and they should be hidden every time you switch Mod?

semhustej, 2025-04-15:
You are right @tetektoza, thanks for pointing that out. I have Draft WB loaded by default, so those widgets always show up. Draft widgets are not available to turn on when the Draft WB is not loaded. Only after that is it possible. 

This seems like a bug to me - users should only have Draft Widgets available in the RMB context menu when the Draft WB is active. The same should be applied for the BIM Widget and the BIM workbench.

I think it is up for debate, whether to enable Draft Widgets also in the BIM workbench. I don't think it is crucial and if it complicates things (which I think it does) I would rather side on the side of not complicating things.

Let's see what the core developers have to say.

semhustej, 2025-06-18:
After thinking for a while my opinion is that both Draft Snap Widget and Draft Scale Widget should be enabled in BIM Workbench.

**Reasoning**

_Draft Snap Widget_
When using BIM workbench, it is quite common to use Draft Snap tools. Some of the Draft Snap tools are actually only BIM-related. Thus it would make sense to enable BIM users the same quick access to Draft Snap tools through the widget as Draft users are offered.

_Draft Scale Widget_
Many tools in the BIM workbench are influenced by the setting of the scale. These include `BIM_Text`, `BIM_DimensionAligned`, `Arch_Axes` and few others. Thus it would make sense to enable BIM users the same quick access to changing Scale as Draft users are offered.

markkomartin, 2025-11-08:
> After thinking for a while my opinion is that both Draft Snap Widget and Draft Scale Widget should be enabled in BIM Workbench.

I agree about the usefulness of the Draft Snap and Draft Scale widgets in BIM Workbench.