FreeCAD/FreeCAD #20787: Part Design: Only one pad is created for multiple closed objects in a sketch created from the origin

ayumu-bekki, 2025-04-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating multiple closed objects in a sketch that originates from the origin, the Pad feature only generates for one of the objects. This issue does not occur when creating a sketch from other faces.

# Steps to Reproduce
1. Create a new sketch on the XY plane (origin).
2. Draw multiple closed objects within the sketch.
3. Attempt to create a Pad feature from the sketch.

[image]
[image]

# Expected Behavior
A Pad feature should be created for all closed objects in the sketch.

# Actual Behavior
The Pad feature is only created for one of the closed objects.

# Additional Information
This issue does not occur when creating a sketch from other faces.

### Full version info

```shell
OS: macOS 14.5
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ayumu-bekki, 2025-04-14:
Sample FCStd File
https://github.com/ayumu-bekki/freecad-bug/blob/main/bug.FCStd

kaiwas, 2025-04-14:
Activate this setting and everything will work.
But thanks to your question I found one incorrectness. I will make another issue

[image]

closed 2025-04-15 by maxwxyz