FreeCAD/FreeCAD #20811: Some constraints in the sketcher with the mouse can be selected while others can't

hansmerkl, 2025-04-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the sketcher I can select some constraints to delete them but others can't be selected. In the attached screen recording I can't select the two constraints on the right side. I assume it's because they are close together

https://github.com/user-attachments/assets/4bec4c7a-89e9-4315-8d92-a2127013d587

[ConstraintSelect.FCStd.zip](https://github.com/user-attachments/files/19759474/ConstraintSelect.FCStd.zip)

### Full version info

```shell
OS: macOS 15.4
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-16:
can you select them when zooming in closer so they appear with a wider gap? Please also test on the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

hansmerkl, 2025-04-16:
@maxwxyz On 1.0 it helps to zoom in, at some point the constraints become selectable.

1.1 shows only one of the two constraints when zoomed out but when you zoom in, it shows both and both are selectable.

Katt-Kushol, 2025-05-06:
[image]

I was able to select/highlight all of the constraints in FC1.1dev (41365 on windows 11). I wonder if the "pick radius" has something to do with this. I have pick radius set up at 5 px.

FEA-eng, 2025-10-07:
Can't reproduce. Can it be OS-specific?

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

jlewin, 2026-08-29:
The original behavior presented in this issue no longer seems to directly repro, as confirmed with the source file on `Windows 11` FreeCAD `1.3` and `main` and as supported by prior reports. In all cases I can select the problem constraint and can do so at all zoom levels and with small (0.5) and large (35) selection radius settings. 

There is one interesting note worth carrying on to related, reproducible variants of this issue and it is that constraint preselection does not seem to honor the selection radius. This is evident when setting the `Selection radius` to its smallest value (0.5) and interacting with the scene. With a miniscule value, picking the axis lines becomes quite difficult, yet has no effect on constraints, which get hit early and easily and from a fixed distance rather than the specified value. Further, at one or more zoomed out views, there was inconsistent picking when moving between constraints which hinted at the original problem and failed to switch to the closest constraint under the mouse. Hard to reproduce and I failed to get a screencast but did observe unexpected picking or at least some stickiness to the current pick. Noting here for reference and will take this knowledge forward, hopefully working toward a better understanding.

jlewin, 2026-08-29:
Some additional notes based on further observations. It seems the viewport scaling has a big effect on picking and candidate selection. Initially in the example below, the constraint is hit as the mouse approaches its bounds, yet after the panel is moved and the viewport is subsequently scaled, and unexpectedly the constraint picking behavior changes. Compared with the intial behavior, it is no longer hit as the bounds are entered and only triggers when we are much farther into its bounds. The quirk depicted here seems highly related to this class of defect.

[image]

jlewin, 2026-08-29:
I can now reproduce the behavior depicted in the original issue. Unexpectedly, the **aspect ratio** of the viewport directly affects the degree of success when picking the target constraint. In the example below, it is impossible to pick the tangent constraint when a tall/skinny viewport is active, yet it works as expected as it gets closer to a square one, or more likely my actual screen aspect ratio. Given that there are a handful of different detection implementations, it also seems reasonable that the `tangent` constraint behavior differs from the sibling `point` constraint for the same aspect ratio.

[image]

FEA-eng, 2026-08-29:
> I can now reproduce the behavior depicted in the original issue.

@maxwxyz Can you reopen it?

maxwxyz, 2026-08-29:
related https://github.com/FreeCAD/FreeCAD/issues/32255 ?
maybe the same fix solves this?

@tritao FYI

tritao, 2026-08-29:
It's a separate issue, thanks for all the info @jlewin, think that's enough to figure out the right fix.