FreeCAD/FreeCAD #20815: Periodic B-spline by knots crashes with more than 2 knots

Cygwin231, 2025-04-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On both the Flatpak version and the version in the Arch extra/ repo, using a periodic B-spline by knots crashes with message: `flatpak[243558]: /usr/include/c++/13.2.0/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.`

I've successfully reproduced this on a fresh Arch install, as well as on a fresh Debian install using the flatpak. The appimage is unaffected, and works everywhere I tested it.

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

Sketcher

### Anything else?

This issue has been brought up before - https://github.com/FreeCAD/FreeCAD/issues/16507 - but was closed by the author after concluding it's an issue of "archlinux or any distro using gcc pre-release". I'm opening this new issue because it's also present in the flatpak.

I was unable to record a macro for it, all my attempts resulted in a macro that simply makes several points in a sketch, without a spline. However, the previous issue had a macro attached with which I was able to reproduce the crash:
`
doc=App.ActiveDocument
doc.addObject('Sketcher::SketchObject', 'Sketch')
doc.Sketch.Placement = App.Placement(App.Vector(0.000000, 0.000000, 0.000000), App.Rotation(0.000000, 0.000000, 0.000000, 1.000000))
doc.Sketch.MapMode = "Deactivated"
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-44.553844,22.012373,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-20.045523,34.417820,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(17.775951,19.440510,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-9.304221,-22.314404,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-38.804981,-11.119246,0)),True)
_finalbsp_poles = []
_finalbsp_knots = []
_finalbsp_mults = []
_bsps = []
_bsps.append(Part.BSplineCurve())
_bsps[-1].interpolate([App.Vector(-44.5538,22.0124),App.Vector(-20.0455,34.4178),App.Vector(17.776,19.4405),App.Vector(-9.30422,-22.3144),App.Vector(-38.805,-11.1192)], PeriodicFlag=True)
_bsps[-1].increaseDegree(3)
_finalbsp_poles.extend(_bsps[-1].getPoles())
_finalbsp_knots.extend(_bsps[-1].getKnots())
_finalbsp_mults.extend(_bsps[-1].getMultiplicities())
doc.getObject('Sketch').addGeometry(Part.BSplineCurve(_finalbsp_poles,_finalbsp_mults,_finalbsp_knots,True,3,None,False),False)
del(_bsps)
del(_finalbsp_poles)
del(_finalbsp_knots)
del(_finalbsp_mults)
conList = []
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',0,1,5,0))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',1,1,5,1))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',2,1,5,2))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',3,1,5,3))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',4,1,5,4))
doc.getObject('Sketch').addConstraint(conList)
doc.getObject('Sketch').exposeInternalGeometry(5)
constraintList = []
doc.recompute()
`

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-16:
could you test if it crashes on the dev appimage? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@AjinkyaDahale FYI

Cygwin231, 2025-04-16:
I am unable to reproduce the crash in either the weekly or the stable appimages. Using the flatpak, steps to reproduce:

1. Make a new part
2. Make a new sketch on any plane
3. Select the "Periodic B-spline by knots" tool
4. Click at least 3 unique points
5. Hit escape, or click on the starting point to finish the spline

Here's an example file, loading it with the appimages works just fine, but loading it with the flatpak causes it to crash.

[RandomPeriodicBSplineByKnots.tar.gz](https://github.com/user-attachments/files/19781601/RandomPeriodicBSplineByKnots.tar.gz)

maxwxyz, 2025-04-16:
thanks for testing.
@adrianinsaval if not reproducible in the weekly appimage but in the stable flatpack it should be resolved, correct?