FreeCAD/FreeCAD #20817: Union with arrays bug in part workbench?

gdvranjes, 2025-04-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you create two arrays as per the attached file and try to make an union of them, it creates an "empty" fusion. It seems related to bugs #17405 and/or #5925.
The problem doesn't occur in Freecad weekly, but needs 30 seconds to compute in my machine.

[array_bug.zip](https://github.com/user-attachments/files/19765346/array_bug.zip)

 

### Full version info

```shell
OS: Slackware 15.0 x86_64 (KDE//usr/share/xsessions/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/52.445/1
Installed mods: 
  * Plot.backup1736919899.3132463 2024.11.26 (Disabled)
  * CurvedShapes.backup1736919830.8936977 1.0.13 (Disabled)
  * AirPlaneDesign 0.4.1
  * Curves.backup1736919830.987689 0.6.53 (Disabled)
  * OpenDark.backup1725671265.1599011 2024.3.13 (Disabled)
  * CurvedShapes 1.0.13
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

gdvranjes, 2025-04-15:
By mistake the previous full version info is for the weekly build I am using without the problem. Below is the 1.0.0 full version info which has the problem...

OS: Slackware 15.0 x86_64 (KDE//usr/share/xsessions/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Plot.backup1736919899.3132463 2024.11.26 (Disabled)
  * CurvedShapes.backup1736919830.8936977 1.0.13 (Disabled)
  * AirPlaneDesign 0.4.1
  * Curves.backup1736919830.987689 0.6.53 (Disabled)
  * OpenDark.backup1725671265.1599011 2024.3.13 (Disabled)
  * CurvedShapes 1.0.13
  * freecad.gears 1.0.0

maxwxyz, 2025-04-16:
I'm closing this as already fixed in the weekly

closed 2025-04-16 by maxwxyz