FreeCAD/FreeCAD #20832: Sketcher CreateSlot slot tool does not automatically constraint points, requiring manually adding constraints

mersenne-twister, 2025-04-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

(this issue may already exist, however is so, I could not locate it here)

in a sketch, when a slot is created, the four points that define the 'corners' of the slot are not effectively automatically constrained, despite there being more than enough information based on what was entered when creating the slot.

in particular, constraints for, eg, horizontal and vertical distance distance constraints for the four corners, from the two starting points, are known, based on the value entered by the user that defines the radius of the curves, etc.

I would expect these constraints to be added automatically based on the rest of the sketching tools, as currently, I have to add at least 4 constraints to every slot.

steps to reproduce: create a sketch. create two points, to my knowledge, their being fully constrained or not should not matter. create a slot based on those two points. select any distance for the size of the slot. the slot will now be under-constrained, despite all the necessary information to constrain it being in the information the user entered.

### Full version info

```shell
freecad 1.0.0 rev. 39109
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-18:
@PaddleStroke FYI
related:
- https://github.com/FreeCAD/FreeCAD/issues/17074
- https://github.com/FreeCAD/FreeCAD/issues/11924

PaddleStroke, 2025-11-17:
@maxwxyz unless I'm missing something please close

PaddleStroke, 2025-11-17:
I think the user probably came upon https://github.com/FreeCAD/FreeCAD/issues/17074 and misinterpreted it.