FreeCAD/FreeCAD #20834: shp.mirror() when shp has a Placement no longer works correctly.

edwilliams16, 2025-04-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Mirroring a shape with _shp.mirror(point, dir)_ gives incorrect results in 1.0+, works correctly in 21.2
The following script illustrates the problem:

```
V=App.Vector
Vx=V(1,0,0); Vy=V(0,1,0); Vz=V(0,0,1); VO=V(0,0,0)
tiltbox = Part.makeBox(10, 20, 30, V(0, 30, 0), V(0,1,1))
Part.show(tiltbox, 'Tiltbox')
tiltbox_mirror = tiltbox.mirror(VO, Vy)
Part.show(tiltbox_mirror, 'Tiltbox_mirror')
tiltbox_placed = Part.makeBox(10, 20, 30)
tiltbox_placed.Placement = App.Placement(V(0, 30, 0), App.Rotation(-VO,-Vx, V(0,1,1),'YZX'))
Part.show(tiltbox_placed, 'Tiltbox_placed')
tiltbox_placed_mirror_bad = tiltbox_placed.mirror(VO, Vy)
Part.show(tiltbox_placed_mirror_bad, 'Tiltbox_placed_mirror_bad')

```
In the above a box is created two ways in the same location, one with an identity placement, the other with a non-identity placement.
When mirrored in the XZ plane they should end up in the same (correct) place.  In 21.2 they do, in the current they don't.

### Full version info

```shell
OS: macOS Sonoma (14.4)
Architecture: arm64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/macos
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * MeshRemodel 1.10.37
  * DynamicData 2.73.0
  * OpenTheme 2024.9.1
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * lattice2.backup1744229912.567353 1.0.0 (Disabled)
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.8
  * fasteners 0.5.38
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.7.22
  * toSketch 1.0.1 (Disabled)
  * Help 1.0.3
  * Alternate_OpenSCAD.backup1744229906.2174761 1.0.0 (Disabled)
  * Curves 0.6.61
```

### Subproject(s) affected?

Core

### Anything else?

See discussion in [https://forum.freecad.org/viewtopic.php?style=10&t=96356](url)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-17:
Quite possibly the regression was introduced when TNP was set to Enabled in May 2024. Last good build I have:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

and first bad build I have is:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

luzpaz, 2025-07-27:
@maxwxyz is this toponaming-related too ?

maxwxyz, 2025-07-27:
IMO not directly toponaming but might have changed with the migration

theosib, 2026-01-15:
Alright. It took claude a while to figure this out, but here are the answers. I hope you don't mine me just posting AI output.

## Answers to Questions

### 1. Is this bug present in LinkStage3?

**The bug was NOT introduced by realthunder.** Looking at the git history:

1. **Commit `8e91aad2f0`** (Feb 22, 2024, by Zheng Lei/realthunder): Added `makEMirror` declaration to TopoShape.h, but **no implementation** was added to TopoShapeExpansion.cpp - only `makESlice` and `makESlices` were implemented.

2. **Commit `dbe7c9d372`** (Feb 22, 2024, by bgbsww): Added the `makeElementMirror` implementation with the buggy placement code:
   ```cpp
   TopLoc_Location loc = shape.getShape().Location();
   gp_Trsf placement = loc.Transformation();
   mat = placement * mat;  // <-- BUG introduced here
   ```

So the bug was introduced during the "cleanups and tests" commit when the implementation was written, not in realthunder's original transfer. It's unclear whether LinkStage3 has a different implementation of `makEMirror` - the declaration was transferred but the implementation appears to be original to the FreeCAD main branch.

### 2. Do we need compatibility handling for 1.0 files?

**No compatibility patch is needed.** Here's why:

The `mirror()` function is a **runtime operation** on shapes - it doesn't persist anything to the file. The mirrored shape result is computed fresh each time from:
- The source shape (stored in the file)
- The mirror axis (parameter to the function)

When a user mirrors a shape:
- **Before the fix**: Wrong result computed at runtime
- **After the fix**: Correct result computed at runtime

The source shapes in 1.0 files are stored correctly - the bug only affected the computation of the mirror result, not what was saved. Fixing the algorithm will automatically produce correct results for all files, old and new.

For `Part::Mirroring` features (the GUI-based Part → Mirror tool), if someone saved a mirrored result, that geometry was computed with the buggy code. However, triggering a recompute after the fix would regenerate the correct geometry. This is expected behavior for parametric features.

### The Fix

The fix simply removes the incorrect placement extraction and multiplication:

```cpp
TopoShape& TopoShape::makeElementMirror(const TopoShape& shape, const gp_Ax2& ax2, const char* op)
{
    if (!op) {
        op = Part::OpCodes::Mirror;
    }
    if (shape.isNull()) {
        FC_THROWM(NullShapeException, "Null shape");
    }
    gp_Trsf mat;
    mat.SetMirror(ax2);
    // Note: Do NOT extract and pre-multiply the shape's Location/Placement here.
    // BRepBuilderAPI_Transform correctly handles shapes with Location already.
    // Pre-multiplying would double-apply the placement, causing incorrect results
    // for shapes with non-identity Placement. See GitHub issue #20834.
    BRepBuilderAPI_Transform mkTrf(shape.getShape(), mat);
    return makeElementShape(mkTrf, shape, op);
}
```

This matches the behavior of the original `TopoShape::mirror()` function (TopoShape.cpp line 3331) which works correctly and doesn't manipulate the placement.

theosib, 2026-01-15:
The bug relates to non-identity placement. Consider two cases. One is we create an object where the geometry is at some location in space, where the offset geometry is all zero. That's identity placement. Next, you could make another object at the origin and then use the placement to move it over. That's non-identity placement.

Now, say you had two objects, one created each way such that they appear coincident in the 3D view. Then we use Part::Mirror on each one.

The mirror of the identity placement one will appear correctly on the opposite side of the mirroring plane (e.g. XZ). But that's because the position adjustment is zero, so applying it twice doesn't hurt anything.

The mirror of the non-identity placement appears in the wrong place because the position adjustment is applied TWICE.

If someone saw this happening in 1.0, they almost certainly wouldn't have used the misplaced result. They would have adjusted their approach to achieve the effect some other way. The buggy result isn't usable or sensible since it puts the mirror image in the wrong location.

theosib, 2026-01-16:
I can easily make a PR for this. I'll have claude code generate it. After I'm done with another PR I'm working on.

closed 2026-01-29 by chennes