FreeCAD/FreeCAD #20836: About FreeCAD - no longer shows complete OS version (macOS at least)

JohnOCFII, 2025-04-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Over the last few months, the format of the Operating System information in the "About FreeCAD" info that can be copied into a clipboard changed from:

`OS: macOS 15.4.1`

to:
`OS: macOS Sequoia (15.4)`

After that change, the full OS data (at least for macOS) is no longer shown.  In the example above, the `.1` is not shown, even though I've just upgraded from macOS 15.4 to 15.4.1

### Full version info

```shell
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.0

OS: macOS Sequoia (15.4)
Architecture: arm64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-17:
@oursland is this a change by FreeCAD or macOS?

oursland, 2025-04-17:
I suspect it is a result of [`src/App/Application.cpp::Application::prettyProductInfoWrapper()`](https://github.com/FreeCAD/FreeCAD/blob/main/src/App/Application.cpp#L3542) in which Qt < 6.5.0 did a manual construction of the OS name for macOS but later versions use `QSysInfo::prettyProductName()`.

oursland, 2025-04-17:
I think in general, the minor release version (i.e. 22.04 for Ubuntu, 15.4 for macOS, etc.) provides a stable API reference.  However, the point releases do sometimes resolve and introduce bugs that affect end-users and software.  I agree that it is beneficial to have the exact version of software release that is installed (i.e. 22.04.3 for Ubuntu, 15.4.1 for macOS, etc.).

PR #20858 (2025-04-18): App: Use the legacy macOS version functionality for About FreeCAD dialog
The Qt version gating for determining OS version for the About FreeCAD dialog resulted in a loss of information. This PR uses the legacy approach which retains the version information details.

Fixes #20836.
closed 2025-04-18 by hyarion