FreeCAD/FreeCAD #20837: About FreeCAD - time to "Copy to Clipboard" lengthy without indication to user

JohnOCFII, 2025-04-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The time it takes to `Copy to Clipboard` averages 12-13 seconds.  Seems long for that operation, and if that operations does require more than a second or two, it should display some indication to the user that something is happening.

### Full version info

```shell
OS: macOS Sequoia (15.4)
Architecture: arm64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-17:
I cannot reproduce on Linux either. In fact, it seems to have gotten quicker in recent times. On my system it takes less than a second for <kbd>Copied!</kbd> to be shown. In the past, it could take 3-4 seconds, but never as much as 12-13 s.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41296 (Git) Snap 1396
Build date: 2025/04/16 22:28:51
Build type: Release
Branch: main
Hash: c8eb23174523efcb5bb6aa89e434382736b7e6c6
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

xtemp09, 2025-04-17:
I have copied too many times to count on _Windows_ and _Linux_, each time was less than 10 ms. _FreeCAD_ has internal functions to measure time. It is easy to use them to write into logs how much time the program spends at any stage.

semhustej, 2025-04-17:
I encountered similar behavior - taking 5-10 seconds to copy FreeCAD info into clipboard - for the past 2 months using the Windows 1.1dev version.
I am not able to reproduce it 100% times, I would guess it's one out of 5 times.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.0

```

Roy-043, 2025-04-17:
I confirm @semhustej's findings. The code to animate of the button may be the culprit.

xtemp09, 2025-04-19:
I have an idea: let's add a progress bar and a status bar to display output. This way, we can easily identify which operations take the longest and optimize them accordingly. Additionally, we might consider parallelizing the copy-to-clipboard function for improved efficiency.

JohnOCFII, 2025-04-27:
I've also noted this is only slow the first time I attempt the `Copy to Clipboard` process.  Once the process completes, if I exit the `About...` panel, and re-enter and try the `Copy to clipboard` process again, it is very fast.  So only slow after the first launch.

Based on some discussion on Discord, I ran with `lldb` and captured the following information, in case this is useful:

Initial data after stopping with `CTRL-C`

```
2025-04-27 12:35:52.002863-0500 FreeCAD[17918:9421597] [CursorUI] ViewBridge to RemoteViewService Terminated: Error Domain=com.apple.ViewBridge Code=18 "(null)" UserInfo={com.apple.ViewBridge.error.hint=this process disconnected remote view controller -- benign unless unexpected, com.apple.ViewBridge.error.description=NSViewBridgeErrorCanceled}
Process 17918 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x0000000182bf5620 dyld`dyld4::JustInTimeLoader::getExportsTrie(unsigned long long&, unsigned int&) const
dyld`dyld4::JustInTimeLoader::getExportsTrie:
->  0x182bf5620 <+0>:  ldr    w8, [x0, #0x70]
    0x182bf5624 <+4>:  cbz    w8, 0x182bf5634 ; <+20>
    0x182bf5628 <+8>:  str    x8, [x1]
    0x182bf562c <+12>: ldr    w9, [x0, #0x74]
Target 0: (FreeCAD) stopped.
```

Data from `BT` command from `lldb` (from a separate launch of `lldb` and FreeCAD.

```
2025-04-27 13:02:10.152111-0500 FreeCAD[18066:9444186] [CursorUI] ViewBridge to RemoteViewService Terminated: Error Domain=com.apple.ViewBridge Code=18 "(null)" UserInfo={com.apple.ViewBridge.error.hint=this process disconnected remote view controller -- benign unless unexpected, com.apple.ViewBridge.error.description=NSViewBridgeErrorCanceled}
Process 18066 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x0000000182c1fe98 dyld`dyld3::MachOFile::trieWalk(Diagnostics&, unsigned char const*, unsigned char const*, char const*) + 256
dyld`dyld3::MachOFile::trieWalk:
->  0x182c1fe98 <+256>: mov    x8, x14
    0x182c1fe9c <+260>: tbz    w11, #0x0, 0x182c1fea8 ; <+272>
    0x182c1fea0 <+264>: mov    w11, #0x1 ; =1 
    0x182c1fea4 <+268>: b      0x182c1feb4    ; <+284>
Target 0: (FreeCAD) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x0000000182c1fe98 dyld`dyld3::MachOFile::trieWalk(Diagnostics&, unsigned char const*, unsigned char const*, char const*) + 256
    frame #1: 0x0000000182bf082c dyld`dyld4::Loader::hasExportedSymbol(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::ExportedSymbolMode, dyld4::Loader::ResolverMode, dyld4::Loader::ResolvedSymbol*, dyld3::Array<dyld4::Loader const*>*) const + 384
    frame #2: 0x0000000182bf0640 dyld`invocation function for block in dyld4::Loader::resolveSymbol(Diagnostics&, dyld4::RuntimeState&, int, char const*, bool, bool, void (unsigned int, unsigned int, dyld4::Loader::ResolvedSymbol const&) block_pointer, bool) const + 196
    frame #3: 0x0000000182bdc040 dyld`dyld4::RuntimeLocks::withLoadersReadLock(void () block_pointer) + 92
    frame #4: 0x0000000182bef898 dyld`dyld4::Loader::resolveSymbol(Diagnostics&, dyld4::RuntimeState&, int, char const*, bool, bool, void (unsigned int, unsigned int, dyld4::Loader::ResolvedSymbol const&) block_pointer, bool) const + 340
    frame #5: 0x0000000182bef6dc dyld`invocation function for block in dyld4::Loader::forEachBindTarget(Diagnostics&, dyld4::RuntimeState&, void (unsigned int, unsigned int, dyld4::Loader::ResolvedSymbol const&) block_pointer, bool, void (dyld4::Loader::ResolvedSymbol const&, bool&) block_pointer, void (dyld4::Loader::ResolvedSymbol const&, bool&) block_pointer) const + 100
    frame #6: 0x0000000182c25640 dyld`invocation function for block in mach_o::Fixups::forEachBindTarget_Opcodes(Diagnostics&, bool, void (mach_o::Fixups::BindTargetInfo const&, bool&) block_pointer, void (mach_o::Fixups::BindTargetInfo const&, bool&) block_pointer) const + 72
    frame #7: 0x0000000182c24b78 dyld`mach_o::Fixups::forEachBind_OpcodesRegular(Diagnostics&, void (char const*, bool, bool, unsigned int, int, unsigned int, unsigned int, unsigned long long, unsigned char, char const*, bool, bool, unsigned long long, bool, bool&) block_pointer) const + 796
    frame #8: 0x0000000182c24634 dyld`mach_o::Fixups::forEachBindUnified_Opcodes(Diagnostics&, bool, void (unsigned long long, unsigned int, mach_o::Fixups::BindTargetInfo const&, bool&) block_pointer, void (unsigned long long, unsigned int, mach_o::Fixups::BindTargetInfo const&, bool&) block_pointer) const + 224
    frame #9: 0x0000000182c23fc4 dyld`mach_o::Fixups::forEachBindTarget_Opcodes(Diagnostics&, bool, void (mach_o::Fixups::BindTargetInfo const&, bool&) block_pointer, void (mach_o::Fixups::BindTargetInfo const&, bool&) block_pointer) const + 208
    frame #10: 0x0000000182bef64c dyld`invocation function for block in dyld4::Loader::forEachBindTarget(Diagnostics&, dyld4::RuntimeState&, void (unsigned int, unsigned int, dyld4::Loader::ResolvedSymbol const&) block_pointer, bool, void (dyld4::Loader::ResolvedSymbol const&, bool&) block_pointer, void (dyld4::Loader::ResolvedSymbol const&, bool&) block_pointer) const + 264
    frame #11: 0x0000000182c2767c dyld`dyld3::MachOAnalyzer::withVMLayout(Diagnostics&, void (mach_o::Layout const&) block_pointer) const + 672
[... cut ...]
  thread #12: tid = 0x901b90, 0x0000000182f2fc34 libsystem_kernel.dylib`mach_msg2_trap + 8, name = 'com.apple.NSEventThread'
  thread #13: tid = 0x901b9a, 0x0000000182f333cc libsystem_kernel.dylib`__psynch_cvwait + 8
  thread #14: tid = 0x901bb5, 0x0000000182f333cc libsystem_kernel.dylib`__psynch_cvwait + 8, name = 'Thread (pooled)'
  thread #16: tid = 0x901bc5, 0x0000000182f318b0 libsystem_kernel.dylib`__workq_kernreturn + 8
  thread #18: tid = 0x901bc9, 0x0000000182f318b0 libsystem_kernel.dylib`__workq_kernreturn + 8
```

semhustej, 2025-04-28:
> Once the process completes, if I exit the `About...` panel, and re-enter and try the `Copy to clipboard` process again, it is very fast. So only slow after the first launch.

This is my experience too.

oursland, 2025-05-07:
That backtrace explains it.  It's doing the just-in-time loading of `ifcopenshell`, which loads a lot of other dependencies.

The offending line is here: [src/App/Application.cpp#L3728](https://github.com/FreeCAD/FreeCAD/blob/8a82caf3aff69c5133801fc3e0fca604c66fbd19/src/App/Application.cpp#L3728)

For the weekly releases, `ifcopenshell` is packaged with the build and does not change.  This is not true for developer builds nor for packager-installed FreeCAD.  We could extract the installed version for the weekly and official builds and use that as a macro and via the Python for other platforms, similar to how the OCCT version is defined.