FreeCAD/FreeCAD #20838: About FreeCAD - Dialog behavior for "Copy to Clipboard" button is inconsistent with typical behavior

JohnOCFII, 2025-04-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When mousing over most interface buttons, the button shading changes to indicate the area is an actual button.  The `Copy to Clipboard` button shows no change in shading when I mouse over it.  Even the `OK` button in the same dialog behaves as expected.

When the `Copy to Clipboard` button is **actually** pressed, shading is changed. (And related to my other recent issue, there is no indication action is actually happening for the 12-13 seconds it takes for the message `Copied!` to be displayed.

[image]

[image]

### Full version info

```shell
OS: macOS Sequoia (15.4)
Architecture: arm64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-17:
I cannot reproduce on Linux and FreeCAD Light. Both buttons are highlighted on hover. Perhaps a macOS-specific issue?

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41296 (Git) Snap 1396
Build date: 2025/04/16 22:28:51
Build type: Release
Branch: main
Hash: c8eb23174523efcb5bb6aa89e434382736b7e6c6
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

maxwxyz, 2025-04-17:
@JohnOCFII is this happening with the 3rd party OpenTheme only? Or also with FreeCAD Dark/Light?

JohnOCFII, 2025-04-17:
> [@JohnOCFII](https://github.com/JohnOCFII) is this happening with the 3rd party OpenTheme only? Or also with FreeCAD Dark/Light?

Great question!

I just switched to FreeCAD Dark and... there is the _slightest_ indication of the background of the `Copy to clipboard` button changing when I hover over it.  Very, very slight difference.

Not hovering:

[image]

Hovering:

[image]

MisterMakerNL, 2025-04-17:
There is even a acknowledgment in my dev branch.
I do see it needs some style-sheet tlc. Buttons are not great. 

[image]

JohnOCFII, 2025-10-06:
This is still an issue in both FreeCAD Dark and OpenDark. (as above, very small shading change in FreeCAD Dark.). This is also an issue in OpenLight, but FreeCAD light is fine.