FreeCAD/FreeCAD #20845: PD: two dimensions option in features is incomplete

69Piero, 2025-04-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

@PaddleStroke 

Currently in Pad/Pocket/Revolution/Groove feature the option "two dimensions" is incomplete and restricted only to a dimensional value. For a complete option you should add in both directions the end conditions "up to face/first/last.." etc.

i mean, in a pad i could need 30mm in one side and "up to face" to the other side. in a revolution 45° in a side and "up to first" to the other side or up to first in both sides

### Full version info

```shell
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * dxf-library
  * freecad.gears 1.3.0
  * FreeCAD_Info
  * symbols_library
```

### Subproject(s) affected?

PartDesign

### Anything else?

@PaddleStroke 

discussed in https://forum.freecad.org/viewtopic.php?t=92443

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-18:
@FlachyJoe FYI.
related:
- https://github.com/FreeCAD/FreeCAD/issues/16175
- https://github.com/FreeCAD/FreeCAD/issues/16175

69Piero, 2025-04-18:
that's my idea, 

"Two dimensions" mode should be renamed into "two directions"

in the "two directions" submenu, both directions (we can name direction 1 and direction 2) must have all option (dimension, up to face/first/last, and through all for the subtractive features).

what do you think?

pierreporte, 2025-05-01:
I think that pad and pocket extrusion type should be decoupled from one direction and two directions. Basically remove two directions from the list and add a radio button. When two directions is selected, then the same form is displayed twice (for each direction). Checkbox for pad symmetric to the base plane should be disabled when two directions is selected.