FreeCAD/FreeCAD #20856: [UI]Missing Qgroupbox in settings menu

MisterMakerNL, 2025-04-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Working on the dark stylesheet I noticed some preference setting pages are missing Qgroupbox, and result in "floating text or floating checkboxes:

[image]

[image]

[image]

[image]

[image]

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MisterMakerNL, 2025-04-18:
I don't think this is an feature but a UI bug also closely related to @obelisk79 styleguide? @FreeCAD/design-working-group 
I also don't think it this hard to fix might even try it myself if I finish the other pr.

behollister, 2025-04-27:
The telemetry preferences ui is part of the Telemetry Addon, and the affected file is:
https://github.com/FreeCAD/FreeCAD-Telemetry/blob/main/Resources/panels/preferences.ui
Not sure if this issue w/ Telemetry should be logged separately there instead?

behollister, 2025-04-28:
Also applies to AddonManager submodule:
https://github.com/FreeCAD/AddonManager/tree/34d433a02c7ec5c73bec9c57d0a27ea70b36c90d

For https://github.com/FreeCAD/AddonManager/blob/34d433a02c7ec5c73bec9c57d0a27ea70b36c90d/AddonManagerOptions.ui

behollister, 2025-04-28:
in addition to pull request, see forum post: https://forum.freecad.org/viewtopic.php?t=96108

hyarion, 2025-04-28:
I think this is rather an issue in the theme.
Group boxes aren’t a requirement in layouts.

Forcing them usually leads to bad/redundant group names.

Group boxes can also be possible to place in other group boxes, so a slight tint might be a better option in the theme.

MisterMakerNL, 2025-04-28:
> I think this is rather an issue in the theme. Group boxes aren’t a requirement in layouts.
> 
> Forcing them usually leads to bad/redundant group names.
> 
> Group boxes can also be possible to place in other group boxes, so a slight tint might be a better option in the theme.

I disagree as it is uniformity.
It is like making up a word document, you go with header1 then header 2 -text.
Header 2 seems to be used as a groupbox header, although if this is correct can be discussed but it seems so for most of the settings.
So how it should like theme wise, is not really the point here, it just should look the same on every page.

hyarion, 2025-04-28:
Just like when writing a word document, if you don't need a header 2, you don't add it.

> So how it should like theme wise, is not really the point here, it just should look the same on every page.

If you want visual consistency with a background panel, then I would just add a panel, without a title.

MisterMakerNL, 2025-04-28:
> If you want visual consistency with a background panel, then I would just add a panel, without a title.

Just that, since it will look more profesional, I know you can hide the Qgroupview, I think Opentheme does it. But classic uses it, and I think it can be useful.