FreeCAD/FreeCAD #20865: [UX] Report view appears if hidden.

kaiwas, 2025-04-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Report view appears if you hide it.

Report view behaves too aggressively. If you disable it in the panel settings, it appears again if it shows an action (for example, if you add a new object).

The problem is especially annoying if you combine report view with the taskbar. During a task, the switch occurs to report view and the focus on the task is lost.

The only way to change this behavior is to completely disable it in the settings. But this completely eliminates the benefit of report view.

In addition, the order of the report view changes when we close the panel in the settings (with the right button).

I would like to have report view always on, but so that it does not draw attention to itself unnecessarily.

Demonstration on video.

https://github.com/user-attachments/assets/fba8d2b2-2309-4692-9d79-0d5633c2c7dd

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda AppImage
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-18:
The option is already present but it seems it doesn't work as expected:

[image]

kaiwas, 2025-04-18:
> The option is already present but it seems it doesn't work as expected:

Yes. It should work in the background. And if I want to see the report view - I will click on the "report view" tab, if it is not visible at the moment.

kaiwas, 2025-04-18:
Report view should work in the background. And if I want to see the report view - I will click on the "report view" tab, if it is not visible at the moment.