FreeCAD/FreeCAD #20871: Inspection tree view icon is from the 90's

ElementW, 2025-04-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image] [image]

Aside from the anachronism, they also do not scale on HiDPI displays (example with 1.5 scaling factor).

[image]

They're even [hardcoded as XPM images in the code](/FreeCAD/FreeCAD/blob/6be09c29f6217c6338cf09dcc9a658454287bd83/src/Mod/Inspection/Gui/ViewProviderInspection.cpp#L761).

### Full version info

```shell
OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-20:
@ElementW Thanks for the ticket. Care to make a proposal on what to change this to ?

ElementW, 2025-04-20:
@luzpaz Not quite sure; either a modern version of the icon based on the Group icon, 
or something more abstract yet representative of what Inspect actually does
[image] [image]

tobiasfalk, 2025-04-20:
> @luzpaz Not quite sure; either a modern version of the icon based on the Group icon, 
> or something more abstract yet representative of what Inspect actually does
> [image] [image]

Maybe the later one, but instead of just the circle, a magnifying glass

ElementW, 2025-04-21:
[image]

Tuned the sizes and colors so it renders better at 16x16

DerRuhri, 2025-05-08:
[image]

Really like the version by @ElementW . Inkscape files indeed are horrible. Exported it in Inkscape as "plain SVG" and had [svgcleaner](https://github.com/RazrFalcon/svgcleaner) and [SVGOMG](https://jakearchibald.github.io/svgomg/) run on it. Also size went down from 15k to 3k. If the result is helpful for you, suit yourself.

ElementW, 2025-05-08:
[image]
2.5k without the needless transforms.

luzpaz, 2025-05-10:
@obelisk79 do you mind weighing-in ?

luzpaz, 2025-05-21:
soft bump for @FreeCAD/design-working-group

obelisk79, 2025-05-22:
I don't find the framing for this issue to be particularly productive. 

There's multiple issues identified. I think the most important is the use of hard coded icons in the tree. These should be SVG so the art can be easily modified as needed. 

Second is a complaint about the art, this is a subjective issue and not an objective one. I generally agree that the current symbol seems to feel 'dated' and the proposed replacement with magnifying glass seems adequate for now despite the fact that I don't understand the significance of the use of rainbow here.

ElementW, 2025-05-22:
> I think the most important is the use of hard coded icons in the tree

That's a technical aspect and is of no interest to end-users, and I didn't create this issue for developer comfort. I created it because the icon is jarring.

> Second is a complaint about the art, this is a subjective issue and not an objective one

All icons are subject to this, I'm not sure why you're pointing this out.

> I don't understand the significance of the use of rainbow here

Way to signal you've never used the inspection WB/tool without saying it.

ElementW, 2025-05-22:
Sigh. Your (XPM) highlight is not *the most important* [of the] *multiple issues identified* yet you made it out to be.

You felt the need to point out the subjectivity of icon design. It's just a fact of UI/icon design. No value in saying it.

Acquainting yourself with what the icon represents would've been the bare minimum before commenting on its appearance. The rainbow is relevant (though not mandatory but probably the best visual cue esp. at small icon sizes) because the inspection workbench turns objects—meshes,  really—into rainbow-coloured triangles denoting the object's similarity to another.

obelisk79, 2025-05-22:
Enjoy your Windows 3.11 folder icon. The distinction was pointed out because any idiot with Inkscape can create an icon, obviously the existing one was created by someone with ms paint. I'm not interested in collaborating with someone who insists on being disrespectful without cause. Someone else from the Design Working Group can deal with such an overtly rude person.

ElementW, 2025-05-22:
I'm acutely aware of the code change necessity (tbh I'm surprised Qt still supports loading inline XPMs like this even in 2025). I don't see why you're trying to make a stink out of it, that's not why I opened the issue in the first place or to discuss; yes the old code will be phased out, end of story, that was implied with its resolution.

I do apologise for the rainbow significance retort. Not for the rest, as your initial response was grating for no reason.

luzpaz, 2025-05-29:
*sigh*

MisterMakerNL, 2025-05-30:
> Enjoy your Windows 3.11 folder icon. The distinction was pointed out because any idiot with Inkscape can create an icon, obviously the existing one was created by someone with ms paint. I'm not interested in collaborating with such a person who insists on being disrespectful without cause. Someone else from the Design Working Group can deal with such an overtly rude person.
> 
> Edit: Of note, the icon you proposed requires a CODE CHANGE in order to support SVG here for it to be used. Kind of an _IMPORTANT_ facet of this and perhaps not as secondary/arbitrary as you want to portray it.

I read the complete thread and I have no idea where this comment comes from. And I think it's very inappropriate.
The person who made the icon was probably more worried about the code and results then the looks. Also since there is no ui department and most dev don't get paid for this, one could understand that not everything can be expected to be of the highest standard.
Also the person did his best to make it usable.

I also have no clue on why a simple icon can cause such frustrations. One could be happy with improvements even if not perfect, many small steps makes a big one.

Further more calling it a rainbow doesn't really shows that you really know what you are looking at, and I really can understand it's kinda hurtful for someone who spend a lot of time on it.

Nevertheless my 2 cents on the icon, in general looks good great improvement.

I'm a bit devided on the magnifying classes as it's commonly used in Freecad for zoom features.
A probe or different measure device that's not a calliper or measuring tape, would be better. But I can understand that this can be difficult to recognize on this scale.

On the rainbow you could do red too green, less colors makes it easier too recognize. And you prevent people thinking it's a rainbow. 😜

ElementW, 2025-05-30:
> I'm a bit devided on the magnifying glass as it's commonly used in Freecad for zoom features

That's a really good point, though as you said other instruments might be harder to recognise. To be fair the Inspection WB icon is a dropper, which I don't even think is appropriately communicating what the WB is about, so there's a longer-standing issue of what to use it seems.

Probes come in all shapes and sizes, and I'm not sure any of them are really evocative especially at small sizes :(

Perhaps a whole different angle to the icon showing the gradient between 2 surfaces intersecting viewed from the side could be something to try?

> On the rainbow you could do red too green, less colors makes it easier too recognize

Yeah it's been a qualm of mine I didn't voice, and why I took the liberty to tweak the colours so the varying luminances wouldn't be so much of a problem. But it's pretty much the only defining feature of an inspection that I can think of—and yes I do think the current colour scheme is bad in many ways. It's a `ColorGradient` with [the default color model pack](https://github.com/FreeCAD/FreeCAD/blob/864449840005def4821bf597e5f3fd062badee91/src/App/ColorModel.cpp#L206), and even though customisable the selections all exhibit major issues, having no hindsight of colour theory or accessibility (think IR imagery or topographic map scales developed to maximise readability), even though [it was reworked 3 years ago](https://github.com/FreeCAD/FreeCAD/commit/38321cc2e90b36c13821d11e986a3e562d510893); a missed opportunity.