FreeCAD/FreeCAD #20879: Freecad 1.0.  cant start

fett-tony, 2025-04-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Fehlerhafter Anwendungsname: freecad.exe, Version: 1.0.0.0, Zeitstempel: 0x673bb0ab
Fehlerhafter Modulname: ucrtbase.dll, Version: 10.0.26100.3624, Zeitstempel: 0x45295404
Ausnahmecode: 0xc0000409
Fehleroffset: 0x00000000000a4ace
Fehlerhafte Prozess-ID: 0x28CC
Fehlerhafte Anwendungsstartzeit: 0x1DBB0CFE92B5C7A
Fehlerhafter Anwendungspfad: C:\Program Files\FreeCAD 1.0\bin\freecad.exe
Fehlerhafter Modulpfad: C:\WINDOWS\System32\ucrtbase.dll
Berichts-ID: caf1a442-bbf4-4d1c-8210-9f32803ab519
Vollständiger Name des fehlerhaften Pakets: 
Fehlerhafte paketbezogene Anwendungs-ID: 

System info:  win11 pro part of a domain!

### Full version info

```shell
cant open while cant start!
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-30:
are you able to start the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2025-05-07:
@fett-tony 👆