FreeCAD/FreeCAD #20881: Std_ToggleClipPlane: provide an additional view

herrdeh, 2025-04-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Std_ToggleClipPlane is an important tool. But with more complex parts it can be difficult to distinguish where is "inside" and "outside" of the part.

Sometimes, the Part_SectionCut view type would come more handy, but it's way slower and always produces new parts in the model.

So I'd like to argue for a control in Std_ToggleClipPlane, which allows to switch to a view which looks like a real cut through the object, with surfaces closed where they would be in reality.

Thanks a lot in advance for your effort,
Wolf

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.12
  * dodo 1.0.1
  * ExtremeProLight 1.6.0
  * Curves 0.6.29
  * BOLTSFC
  * nurbs
  * animation
  * IconThemes
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-19:
Duplicate #5570

herrdeh, 2025-04-20:
> Duplicate [#5570](https://github.com/FreeCAD/FreeCAD/issues/5570)

Agreed. Thank you.

closed 2025-04-20 by herrdeh