FreeCAD/FreeCAD #20882: [BIM][FEATURE] Changing window sill parameter after window created

ErtugrulEge, 2025-04-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you create a window there is a property called sill height which defines the height from gorund of window but after creating window, you cant access this property and you should make some move operations. 

In property view data tab, we can make a "sill height"  property. 

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-04-19:
Sill height in the interactive tool just meant to place the Window/Base Sketch Placement.z

kaiwas, 2025-04-19:
> Sill height in the interactive tool just meant to place the Window/Base Sketch Placement.z

Yes. But what prevented us from taking this property to the level of the "Window" object? After all, the frame width and other dimensions involved in the sketch are on the window object.

It is terribly inconvenient to look for this parameter on the sketch. It is enough to change the height of the window sill for 10 windows to understand that it was difficult to come up with a solution.

closed 2025-05-13 by yorikvanhavre

paullee0, 2026-01-05:
After reverting previous PRs, I think the implementation of this to fit people's intuition is\to mandate a Sill constraint in the Window Base Sketch.

Without this (need properly define), Sill Height property in Window object should not function then.

Any other idea ?