FreeCAD/FreeCAD #20883: Bug  Protusion with a specific sketch

remylabene, 2025-04-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Protusion does not work with a particular sketch scheme, see file joint and the sketch001. If you change the constraint 30 mm with 29 mm, is good, or if you create the same sketch not in XY plane is good. 
See file Bras-01.FCStd (bug) and Bras-02.FCStd (good)
links files : 
https://e.pcloud.link/publink/show?code=XZii7qZP6pvkyhvehY3gsnUr58Teu6WF1aV
https://e.pcloud.link/publink/show?code=XZXr7qZUSuBfDixC8firpOgbybWV5UtnNzy

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * sheetmetal 0.7.22
  * MOOC 2022.4.21
  * MeshRemodel 1.10.37
  * InventorLoader 1.5.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-04-19:
"Protrusion" is the name of the tool in French, in English it's "Pad"; please update the issue title and text accordingly. 
Description is hard to understand, what did you expect and what do you get instead? Does the Report view state anything?

The Bras-01 file you provided just has its geometry out of date yet incorrectly considered up to date, but forcing a refresh (see corresponding command in the right click menu of the pad/protrusion) or editing the length of the pad creates the proper geometry.

Can't reproduce your issue on either 1.0.0 or 1.1.0dev AppImage.

remylabene, 2025-04-19:
I rebuilt this mechanical part, it seems to work, it must have just been a bad database record. I'm closing the bug

remylabene, 2025-04-28:
Another bug, see the video : https://e.pcloud.link/publink/show?code=XZTOmqZJALmcgBwPJL0M3zA0dSp2kUDfU3y

ElementW, 2025-04-28:
This new problem of yours is not related to the Pad tool. Please close this, search if your issue already exists, and only if it doesn't you can open a new one.  
If you do so, please include reproduction steps in text form, not as a video with epilepsy-inducing glitches, and provide a copy of the FCStd file you have.  
Upload files to GitHub directly by drag and dropping your files on the text field, avoid using third party hosting sites as these may go down at any moment or block people/regions/countries without notice.

remylabene, 2025-04-29:
I try this with FreeCad 1.1 build on Linux Debian

[ModelisationPiecesFreeCad-0.0.20.pdf](https://github.com/user-attachments/files/19959149/ModelisationPiecesFreeCad-0.0.20.pdf)

ElementW, 2025-04-29:
Can reproduce on latest weekly.
<details>

```
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.8.1.2
  * SearchBar 1.0.2
```

</details>

In addition, you are creating a sketch with a line that ends up with 0 length, though I didn't notice any influence from this on the bug (also happens without). You creating a 0-length line is a UX issue and warrants its own issue as this goes unnoticed way too easily, and the message in the Report view is not actionable (it mentions an Edge, but edges and Sketcher primitives can not be correlated by users).

You know, providing the FCStd would've been ***really*** useful from the start, instead of having to consume a 3 minute video.

[not_padding.FCStd.zip](https://github.com/user-attachments/files/19964852/not_padding.FCStd.zip)

As for your specific set of issues
* The broken pocket is likely #15452
* The broken pad *could* be #18248, but I observed no error

Definitely an https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC issue either way. @pieterhijma could you take a look?

remylabene, 2025-04-29:
It's the same bug

[PieceTestPad-0.0.1.FCStd.zip](https://github.com/user-attachments/files/19966567/PieceTestPad-0.0.1.FCStd.zip)

closed 2025-04-30 by maxwxyz

pieterhijma, 2025-05-01:
> Definitely an [ 3rd party: OCC](https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC) issue either way. [@pieterhijma](https://github.com/pieterhijma) could you take a look? And maybe identify if the provided file matches those? You can also see weirdness happening at 2:38 in the video.

Thanks, I'll pick it up automatically when I get to the duplicate issue.