FreeCAD/FreeCAD #20884: Varsets: Displayed / Calculated length bugged when referencing same varset

Andersama, 2025-04-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When attempting to add a property which should be calculated using a combination of other properties from the same set the result will be 0. However, the preview calculation will be the correct value. When accepted the leading `<<Varset>>.` or `Varset.` part of the expression is dropped.

This behavior may be intended to prevent self-referential/infinite looping calculations, however it is a bit unintuitive.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-21:
@pieterhijma could you take a look at this?

pieterhijma, 2025-04-21:
@Andersama Could it be that this is a duplicate or at least related to https://github.com/FreeCAD/FreeCAD/issues/19698?

pieterhijma, 2025-06-02:
Ok, thanks @Andersama . I agree as well. @maxwxyz, I noticed you already marked it as duplicate, can you also somehow link them?