FreeCAD/FreeCAD #20886: BIM: BIM Views' Isolate command unexpectedly hides all levels

furgo16, 2025-04-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

With the BIM Views panel active:

1. Select a level in the panel
2. Right click to show the context menu
3. Select the "Isolate" command
4. **Expected**: all levels but the one selected are hidden
5. **Actual**: all levels including the one selected are hidden

As a workaround, double-clicking on a level seems to do what Isolate would be supposed to do. However, my expectation would be that double-clicking activates the level and does not hide anything (but that'd be a separate issue).

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41316 (Git) Snap 1401
Build date: 2025/04/18 22:24:46
Build type: Release
Branch: main
Hash: f60a0a39f299e2704d1b20f728098c2ad870c830
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-04-20:
I can confirm the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0

```

furgo16, 2025-04-25:
Thanks for testing. I've updated the description with a brief video to illustrate the steps to reproduce.

tetektoza, 2025-04-25:
Interesting; but it seems I've done the same and it behaved correctly, no? Can you provide your project file as well?

furgo16, 2025-04-25:
@tetektoza I've added a link to the simple BIM file with three levels I used for the video in the description.

I don't quite follow all the steps in your video, but the one thing I noticed is that in the first 10 s the end result is the same as in the issue, but the steps are slightly different: generally when clicking on a level on the BIM views panel, it takes a moment to see the level selected in the tree view. In your case, you right-click before the level gets selected in the tree view, and then click on Isolate. Just a small detail, which I don't know whether it makes a difference. The end result is still all levels being unexpectedly hidden.

As mentioned in the description, for me the only way to make it behave as expected is to double-click on the level in the views panel. Only then the level is shown and all others are hidden. I would probably define the double-click action as "Activate" instead of "Isolate", but that would be the subject for another issue.

kaiwas, 2025-04-25:
I think we need to think carefully about how levels, groups and the work plane behave. The current implementation is very, very, very confusing. It is not workable. Maybe it is worth creating a separate discussion on this topic?

furgo16, 2025-04-25:
> Maybe it is worth creating a separate discussion on this topic?

Makes sense.

semhustej, 2025-04-25:
Thanks for looking into the issue @tetektoza , I did some tests and found out, that the problem that @furgo16 reported occurs ONLY when Levels are placed in a `Building` container (which is a proper way to structure BIM project). See behavior in the video:

https://github.com/user-attachments/assets/570cef02-4f2c-445d-9eae-624d97d9f02f

Here is a sample file with levels with no Building objects, isolating levels works fine: 

[levels-testing.FCStd.zip](https://github.com/user-attachments/files/19905553/levels-testing.FCStd.zip)

Here is a sample file with levels in a Building objects, isolating levels does not work fine:

[building-levels-testing.FCStd.zip](https://github.com/user-attachments/files/19905575/building-levels-testing.FCStd.zip)

semhustej, 2025-04-25:
FYI This is a typical BIM file with Site -> Building -> Levels spatial structure as a showcase that Levels might be in multiple layered containers: 

[site-building.FCStd.zip](https://github.com/user-attachments/files/19905625/site-building.FCStd.zip)

[image]

semhustej, 2025-04-25:
> > Maybe it is worth creating a separate discussion on this topic?
> 
> Makes sense.

Yes, let's leave this one just to the original `Isolate` issue.

tetektoza, 2025-04-25:
@furgo16 @semhustej FYI - I've analyzed how `isolate` behaves, and it seems what it is doing now is it basically hiding the parent of the childs. You can achieve the same behavior as furgo if you put one level into another, and hide the child level. And we can see the same on furgo's recording - it's not that other levels are being hidden - the other levels are being hidden because we prioritize on hiding parent (which is Building), which in turn hides all of it's childs. Solution is easy - basically, do not hide based on parent of the selected item, but hide based on selected item and anything that is it's child.

I can post a PR, but from user perspective I'm not too advanced in BIM. Do you have any idea why it could be implemented this why that I've described above?

semhustej, 2025-04-25:
> [@furgo16](https://github.com/furgo16) [@semhustej](https://github.com/semhustej) FYI - I've analyzed how `isolate` behaves, and it seems what it is doing now is it basically hiding the parent of the childs. You can achieve the same behavior as furgo if you put one level into another, and hide the child level. And we can see the same on furgo's recording - it's not that other levels are being hidden - the other levels are being hidden because we prioritize on hiding parent (which is Building), which in turn hides all of it's childs. Solution is easy - basically, do not hide based on parent of the selected item, but hide based on selected item and anything that is it's child.
> 
> I can post a PR, but from user perspective I'm not too advanced in BIM. Do you have any idea why it could be implemented this way that I've described above?

**The person who will know for sure is @yorikvanhavre . Let's wait for his input.**

**This is my point of view:**
I can not think of a workflow where this would be desirable. I think it's a bug. If I wanted to isolate the building, I would just choose the building and isolate that. 
I think the bug came from the thought that isolate means hiding 'everything else' but the selected level. But the 'everything else' includes also the building, which in turns off the layer.

There is also a case where two buildings are on in one site, but I think this should not create problems as Building B is not a child of level in building A. I went ahead and created a sample file to be able to test how the functionality works:

[site-2xbuilding - Copy.FCStd.zip](https://github.com/user-attachments/files/19909066/site-2xbuilding.-.Copy.FCStd.zip)

[image]

**I think the functionality should be:**

- Isolating a Level 
*should hide all other levels and the parent Building object.
*Also other building/levels in the project?? I am not sure about this but it would make sense to me - I want to isolate just the selected level.
*if there is a level or group inside of the isolating level, these should not be hidden as they are children of the level
*currently if object inside of the Level (child of the level) is hidden, it stays hidden after the level is isolated. I think this behavior should stay as it is.

- Isolating a Building object 
*should hide all other buildings/levels and children of these other buildings

- Currently objects which are not inside any building are not touched by the actions in the BIM views panel. I would leave this as is.

There are many corner cases that come into my mind but this seems like a good base case to me.

furgo16, 2025-04-25:
To keep things simple, I think the behavior of right-clicking a level on the BIM Views panel and choosing Isolate should be the  same behavior as what currently happens when double-clicking a level in the BIM Views panel.

That is, all levels are hidden except for the one double-clicked on.

closed 2025-05-13 by yorikvanhavre