FreeCAD/FreeCAD #20889: STEP export

violettinged, 2025-04-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I noticed an issue with STEP files containing fillets which are created in FreeCAD. Everything looks fine in FreeCAD, but these STEP files are read incorrectly (the create a rounded edge instead of a fillet) by some CAD programs (Onshape, Rhino), while some other programs apparently have no trouble with these STEP files (Fusion, FreeCAD itself). The geometry is a boolean union of a Box and a cylinder and a fillet in between. The cylinder is either created by the extrude or revolve command. Depending on the "sign" of the extrusion direction or the revolution aixs, the STEP files are read correctly or not.

The entire problem only occurs in v1.0.0, I could not reproduce it in v0.21.1 or v0.20.2.

[image]

### Full version info

```shell
OS: macOS 14.7.4
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-20:
Make sure you're not reporting a duplicate: https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20Fillets

maxwxyz, 2025-04-21:
Please share the file as .zip and check if it works on the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@pieterhijma FYI, it seems to work in older versions.

pieterhijma, 2025-04-21:
Strange issue. I'll look into it. @violettinged, could you share a little bit more detail? You created a cube by extrusion, but what was the surface you extruded from and to which was it attached? The top of the cube? A file (or multiple) would also be nice.

violettinged, 2025-04-21:
I tested 1.1.0dev.41314 (Architecture: x86_64, not arm64!), the problem is also present there. Please find attached the FCStd file (just remove that .zip extension from the file). The features giving wrong STEP files are named accordingly.

[issue_STEP_exp_fillets.FCStd.zip](https://github.com/user-attachments/files/19831757/issue_STEP_exp_fillets.FCStd.zip)

violettinged, 2025-04-21:
The issue discussed here is likely related to #14710. The issue discussed here triggers the flipped spherical surfaces with the extrusion (inverted extrusion direction) or revolution (inverted revolution axis) command, in the referenced issue they are triggered with the sweep command (inverted profile sketch orientation).

I can also confirm that IGES format is not affected in the issue discussed here.

Note that for the issue discussed here, I couldn't reproduce it in versions <1.0.0 (at least not for this minimalistic geometry). In contrast, the forum discussion related to the referenced issue #14710 says, that it is also present in version <1.0.0.

UR-0, 2025-04-22:
Looks like the same: [https://github.com/Open-Cascade-SAS/OCCT/issues/378](url)

bisonactual, 2026-09-06:
Regression was introduced in https://github.com/Open-Cascade-SAS/OCCT/commit/9b9aac4a7b420d6be1e59e1ac9795647fa0acf64 - FixReorder(true) flipped the [circular] boundary edges after Directfaces converts the coordinate system. disabling pcurves fixes it, but likely introduces old torus bugs by disabling the combined ordering this [assumedly] was meant to fix. 

....a few hours after writing the top part of this comment. The reworked algorithm was fine, but the exporters input forced a force wire reverse prior to the good stuff. Fix is trivial, but gotta test its not broken anything.

stefano2734, 2026-09-12:
> Regression was introduced in [Open-Cascade-SAS/OCCT@9b9aac4](https://github.com/Open-Cascade-SAS/OCCT/commit/9b9aac4a7b420d6be1e59e1ac9795647fa0acf64) - FixReorder(true) flipped the [circular] boundary edges after Directfaces converts the coordinate system. disabling pcurves fixes it, but likely introduces old torus bugs by disabling the combined ordering this [assumedly] was meant to fix.
> 
> ....a few hours after writing the top part of this comment. The reworked algorithm was fine, but the exporters input forced a force wire reverse prior to the good stuff. Fix is trivial, but gotta test its not broken anything.

In what release was or will this implemented: occt 8.0.1, 8.0.2, 8.1.0 or later?