FreeCAD/FreeCAD #20895: CAM workbench: Parentheses in tool labels are not escaped/replaced in G-code comments

kalbfuss, 2025-04-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the CAM workbench, parentheses in tool labels are not escaped/removed during G-code generation, but are transferred to respective comments. At least my GRBL controller is not able to interpret the resulting double parentheses and assumes the comment ends early, which produces an error message (best case) or erratic movement.

Expected behaviour: All unsafe characters are escaped/removed in the G-code to avoid error messages/misinterpretation of comments as G-Code commands. Parentheses in tool labels could for instance be stripped.

Steps for reproduction:

1. Create a job in the CAM workbench.
2. Add a tool and name it "TC: 1/8" Endmill (fast)".

[image]

4. Add an arbitrary operation.
5. Use the "Post Process" function to generate the G-Code.
Excerpt from resulting G-Code:
...
(Begin operation: TC: 1/8" Endmill (fast))
...

Error message thrown by GRBL:
> (Begin operation: TC: 1/8" Endmill (fast)) (line=11)
error:1 (Expected command letter)

### Full version info

```shell
v1.0.0 snap package run under Ubuntu Linux 24.04
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-21:
Please add full about info. If possible, please test on the weekly dev version.

LarryWoestman, 2025-04-21:
As I understand the postprocessor code base, the problem description is correct as far as it goes.

Removing "unsafe" characters has two problems:

What is an "unsafe" character varies from controller to controller.

Removing characters from a label may not produce a "valid" comment either.  I'm not sure if replacing "unsafe" characters with "_" or " " would always work.

If you are going to create labels that aren't "safe" for your controller you are asking a lot of the postprocessors to "fix" the problem.

kalbfuss, 2025-04-21:
As a user I do not necessarily know that you use those labels in G-code comments. If everything goes well users will never have to look at the G-code. Unfortunately, G-code comments are delimited by parenthesis. So in my view, stripping all parenthesis from labels before inserting them into G-Code comments makes the G-Code generation a lot more robust. Apart from that you claim to support GRBL controllers (at least it is an option I can select). Would it not make sense to avoid G-Code that causes errors on GRBL controllers?

LarryWoestman, 2025-04-21:
Turn it around a bit:  why would you expect that parenthesis are "safe" to use in labels to begin with?

I believe that labels can be used in expressions, so a lot of characters would be "unsafe" to use in labels, probably including parenthesis.

If you limit labels to the same characters that are valid in computer language variables (letters, numbers, and "_") then you should have "safe" gcode comments and expressions as well.

The "comment characters" for gcode varies from controller to controller.  Parenthesis are relatively common, as is a leading ";", a leading "#", and I think I have seen a leading ":", and leading and trailing '"'.  I doubt I have seen them all.

Rather than trying to "fix" "unsafe" labels in the postprocessors and expression handler, perhaps it makes more sense to limit labels to "safe" characters to start with?

LarryWoestman, 2025-04-21:
To the best of my knowledge, gcode doesn't have any way to "escape" characters.  It is a fairly "primitive" language that was largely invented over 50 years ago and hasn't evolved much since then.

kalbfuss, 2025-04-21:
I am definitely not a G-code expert. That is why I wrote "escaped/replaced". You could also add "stripped". Limiting the content of labels is another approach. The question is, which characters would be safe under all circumstances. That question seems a lot more difficult to answer to me.

The postprocessor should know which syntax is used for comments in the selected dialect. If I am not mistaken, it creates and inserts those comments. Should it not also ensure that those comments do not create error message with controllers using the selected dialect? I mean it is not very surprising that a ")" in the comment causes trouble if the comment is delimited by a ")".

Up to you how to proceed. I am aware of the limitation now and understand how to avoid it. This is more about other users who may run into the same issue.

JohnOCFII, 2025-04-21:
Just adding a comment.  As a former developer (years ago!) we were always expected to sanitize any user free-form input as required for downstream use.  If there is a subset of characters that are known to be safe, I'd think the best option would be to sanitize at input time, and give feedback to the user if they attempted to use "unsafe" characters.  I've run into this with tools in CAM as it accepted characters in my tool name that were then later used to create actual file names in the file system -- and the ensuring file name was invalid in macOS.  I thought I'd opened a previous Github issue related to this, but after paging through the open CAM related issues, can't find it now.

JohnOCFII, 2025-04-22:
@kalbfuss 
> At least my GRBL controller is not able to interpret the resulting double parentheses and assumes the comment ends early, which produces an error message (best case) or erratic movement.

What GRBL processor are you using?  I did a little research, and it seems clear that in some GRBL processors, parentheses are used as delimiters, which means that parentheses used _inside_ of the comment (as in your case) would cause an error.  I'm trying to find a definitive list of acceptable characters, but to date this seems like a good, safe list:

- Letters
- Numbers
- Spaces and tabs
- This list of punctuation: . , : ; ' " ! ? - _ / = + * # $ % & @

Frankly, I'd tend towards allowing fewer of those punctuation items, if we felt it would be safer across more processors.

LarryWoestman, 2025-04-22:
"." has special meaning in Windows systems when used in file names.

There are similar issues with almost all of the characters in your list of punctuation.

Do we want to give people the ability to use the characters that work for their operating system and controller but won't work for some other operating system or controller?  If we limit ourselves to the characters that work "everywhere" we have a pretty short list.  If we don't we will have many issues and questions submitted by people who don't understand the "rules" for their particular combination.

LarryWoestman, 2025-04-22:
Assuming that we opt for limiting labels to the "short list", we are proposing to affect code wherever labels can be entered, and we have an issue with supporting older FreeCAD files where the rules were looser.

It seems to me that we would need to get a wider range of feedback on this level of change.

It might be "simpler" to go ahead and strip/substitute "unsafe" characters in gcode comments (assuming that is the only place that would be affected and that it would be possible to do it) rather than make changes in many other places.  I'm less fond of that solution but it might be the "least bad" overall solution.

That would still leave the possibility that labels might contain "unsafe" characters for other situations such as expressions and file names, at least.  I assume that if such problems existed, that they would already have been noticed and dealt with.  That might be a bit too much of an assumption :-)

JohnOCFII, 2025-04-23:
Agree that the topic needs more input.  Short term, sanitizing before the user-entered strings before they become comments might be the way to limit scope.

Longer term - needs more review -- I had hoped @obelisk79 might have time this summer to help review the CAM Workbench from a User Interface Point of view.

obelisk79, 2025-04-23:
Maybe, got a bit of a busy summer planned. Just idly harass me about it every few weeks I guess.

kalbfuss, 2025-04-23:
@JohnOCFII Not sure if this is still relevant, but here is the version information retrieved from my controller:

```
Grbl 1.1f ['$' for help]
Grbl for ARM32
Version:ARM32 V3.4

> $I
[VER:ARM32 V3.4.20240825:]
[OPT:VZL,35,254]
ok
```

It is the default controller, which comes with the Genmitsu 4040 Pro Max CNC router (bought in 2025).

JohnOCFII, 2025-04-25:
> That would still leave the possibility that labels might contain "unsafe" characters for other situations such as expressions and file names, at least. I assume that if such problems existed, that they would already have been noticed and dealt with. That might be a bit too much of an assumption :-)

@LarryWoestman - I found the older issue with tool names creating file names that may or may not be valid for that OS.  It looks like you commented at the time too.  I'm not sure of the best way forward, but just linking this here as it is somewhat related to the larger issue of user-entered strings being used without proper sanitation. https://github.com/FreeCAD/FreeCAD/issues/11522

LarryWoestman, 2025-04-26:
That does point out that the problem includes file names as well, and may not be already dealt with :-(

It would be nice to have the ability to write a label with little to no concern for "unsafe" characters, but have expressions, file names, and gcode comments (at least, there might be other cases) "do the right thing" to make sure that unsafe characters don't break anything.  That would require at least one function/method/routine that would replace "unsafe" characters with "safe" characters.

A down side to doing that is that file names, expressions, and gcode comments wouldn't match the original label very well, causing some amount of confusion.

As stated before, another possibility is to force labels to meet the "no unsafe characters" requirement right from the beginning.  The label is less expressive, but the resulting label matches the expressions, file names, and gcode comments, leading to less confusion, I hope.  That would still require dealing with labels, expressions, file names, and gcode comments that were created under older, less restrictive rules.

At the moment I'm leaning to the using last option.

JohnOCFII, 2025-04-26:
Yeah, I don't know if there are any easy answers.  I'd agree that given the current situation, the last option seems best, and most maintainable to me.

With hindsight, it seems that having a single database maintaining all the tools information might have been a good solution, but that's just Monday morning quarterbacking from someone who's development skills are very thin.

LarryWoestman, 2025-04-26:
It seems to me that we have multiple things to think about:

Figure out what the "most correct" answer is for labels and the fact that labels show up in some form or other in expressions, file names, and gcode comments (at least).  This needs more input than just two or three people.  How do we get such input and/or who should be making such a decision for FreeCAD?

Figure out how much change is required to implement the "most correct answer".  We might compromise and settle for a "more possible" implementation in the short term if the amount of change is too much to do or involves too much risk.  This could turn into a fairly involved effort to fully implement.

How do we handle labels, etc. from older versions of FreeCAD?

What should our next step(s) be?  I think we are at the point where we have talked about this as much as we can and it needs to go to the appropriate wider audience.

JohnOCFII, 2025-04-26:
@knipknap - wanted to draw your attention to this issue thread ( as well as the other thread linked above), since you are doing some heavy thinking around ToolBits.

knipknap, 2025-04-27:
My opinion: The tool labels are intended for the UI - for human friendly, search friendly presentation. Limiting the UI because a downstream machine representation does not support some characters is not the right thing to do.

What if tomorrow a machine doesn't support numbers? Keep in mind that Gcode is only one of many languages that CAM may need to be translated to.

IMO it can only be postprocessor responsibility to come up with an escaping that works. Everything else just isn't general enough.

Ad yes, there is always an escaping that works. That is Shannon's Source Coding Theorem.

JohnOCFII, 2025-04-27:
> My opinion: The tool labels are intended for the UI - for human friendly, search friendly presentation.

The other issue linked about and now here (https://github.com/FreeCAD/FreeCAD/issues/11522) is about the labels being used as the basis of file names in the operating system with similar issues popping up.  It makes me think that instead of these individual tools and files relying on the operating system individually, they should all be in some sort of single tool library database - but guess that THAT is a ton of work.

knipknap, 2025-04-27:
Yep, after the BTL integration the tool file names will use UUIDs for that reason (also to keep the road to a tool database open).

LarryWoestman, 2025-04-27:
In my opinion any needed file names should not be "derived" from the label anyway.  We should ask the human what filename (or filenames) they want to use.  At most we could provide a file-system-safe version of the label as a starting value for the filename.  Removing characters from the label or replacing them with "safe" characters, then possibly limiting the total number of characters in the starting value for the filename may be needed.

@knipknap I'm not sure what you mean by "escaping".  Gcode doesn't have any way to say "ignore the following character".  You have the option of deleting any "unsafe" characters or replacing them with "safe" characters, as far as I can tell.  Both leave you with something that might not look much like the original label, which could be a source of confusion but may also the best that we can do for gcode.

knipknap, 2025-04-27:
@LarryWoestman you are right, escaping is the wrong term. I meant "encoding". I.e. substitute unsafe characters by (sequences of) safe ones. The fact that they don't look like the original one thereafter is unavoidable, of course.

About the coming change in handling filenames: the user may still rename the file, but BTL won't ask and use UUIDs by default. Because otherwise the semantics get unclear AND all references break as soon as someone changes a label. It's just bad design to use labels as references.

LarryWoestman, 2025-05-31:
In my opinion there are several overlapping issues that have been discussed in this issue:

1)  As things are currently written, it is possible to create tool labels that can contain characters that will cause problems later for certain controllers who don't like gcode comments (at least) that contain those characters.

2) As things are currently written, it appears to be possible to create tool labels that contain characters that are invalid when used as part of a file name.  Which characters are invalid may vary depending on which operating system you are using to run FreeCAD.  It appears that this issue has been mentioned before but is not resolved in the current code base.

3) As things are currently written, it may be possible to create tool labels that contain characters that are invalid when used as part of an expression in FreeCAD.  This possibility needs to be verified.  It may be that case that expressions can handle any characters that the tool labels allow.

4)  Already existing FreeCAD files contain tool labels with characters that might cause the above listed issues.  Any solution needs to be able to be able to accept such files and do something reasonable with them.

LarryWoestman, 2025-05-31:
In my opinion, there are at least a couple of proposed fixes for the issues described above:

Proposed fix 1:

Change the definition of what characters are valid for tool labels (at least) to a set of characters that is valid for gcode comments, file names (in all three operating systems), and expressions.  This proposal has a very broad scope, with effects all over the FreeCAD code base and the changes would also be visible to the uses of FreeCAD.

It has the advantage that it would fix issues 1-3 that I listed above.

It has several disadvantages:

It requires many changes to the code in many areas of FreeCAD.  This would require more than one PR.

It changes the UI and the way that FreeCAD works, so would need a very careful and broad review by many interested parties (including the UI and CAD working groups as well as other interested parties).

It doesn't address issue 4.  We still need an answer for how to handle older FreeCAD files who have tool labels that contain what are now invalid characters.

Proposed fix 2:

Change some or all of the postprocessors to replace any "invalid" characters with "_" or " " in gcode comments.  Leave it up to others to deal with the issues related to tool labels with characters that might cause problems in file names in some operating systems.  Leave it up to others to deal with any possible issues related to tool labels in expressions.

It has the advantages of being of much smaller scope and should be able to fix issue 1 and the part of issue 4 that is related to tool labels showing up in gcode comments.

It has several disadvantages:

Tool labels in gcode comments will be different than the original tool label which might cause confusion and support questions.

Defining what are "invalid" characters in gcode comments is hard and we probably don't know all possible cases.  The one case that I am reasonably sure of is that we don't want whatever character(s) indicate a gcode comment to be able to show up "inside" of a gcode comment.  For example, if comments are indicated by a ";" at the beginning of the line, then we don't want to see more ";" in the middle of the comment.  Another example that started this discussion is a tool label that contained "(" and ")" showing up in a comment that was indicated by a "(" at the beginning of the comment line and ended by a ")" at the end of the comment line.

What are "invalid" characters is highly likely to vary depending on the gcode "dialect" be used to talk to various controllers, and might even vary depending on the version of the software running on the controllers.  This implies that any solution needs a way to specify a list of the "invalid" characters at least in the code for the postprocessor and possibly from a "command line argument" for the postprocessor.

Defining what character (or characters) to use to replace any "invalid" characters needs to be done.  "_" is my first choice, but " " might also work.  This implies that any solution needs to specify what character (or characters) should be used at least in the code for the postprocessor and possibly from a "command line argument" for the postprocessor.

The issues related to tool labels with characters that might cause problems in file names and expressions would still need to be fixed.  At minimum additional "FreeCAD github issues" should be filed describing these issues.

Then there is a meta-issue :-)

Who makes the decision about which way to fix these issues?

LarryWoestman, 2025-05-31:
I feel comfortable proposing that we choose fix 2.  That is of narrow enough scope that it would be a "good start", and the rest of the issues can be filed for future consideration.  It is unlikely to be controversial or need discussion by many interested parties.

LarryWoestman, 2025-05-31:
One additional thought:  It is helpful to allow "rich" or "expressive" tool labels.  Limiting the tool labels to just letters, numbers, and "_" (for example) would remove a lot of the ability to describe a tool.

LarryWoestman, 2025-05-31:
@maxwxyz  or @luzpaz would you please assign this issue to me?  I don't appear to have the ability to do this myself.

Thanks!

JohnOCFII, 2025-06-01:
> I feel comfortable proposing that we choose fix 2.

Good analysis.  I think that the Tool Library work that @knipknap is working on will help resolve some of these issues (like the file system issue).  Hopefully they'll chime in.

knipknap, 2025-06-01:
@JohnOCFII I think the fundamental problem will remain, even with the new tool library. Though it is making progress towards using IDs for tools that are not generated from labels, which could eventually allow to use databases instead, making files obsolete. But I support the approach proposed by @LarryWoestman , option 2 is better IMO.