FreeCAD/FreeCAD #20898: two points on a circle crossing an arbitrary axis?

petaflot, 2025-04-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

constraining a point on an axis and a circle reports "fully constrained" but this is wrong

[caca.zip](https://github.com/user-attachments/files/19837287/caca.zip)

there are (in most cases, and definitely in this case) obviously two points on the circle that match the constraints

clue: use quaternions

### Full version info

```shell
1.0.0
```

### Subproject(s) affected?

None

### Anything else?

this is going to hurt

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-04-21:
I would like to add, if this is not an issue that's already tracked, that this possibility of 2+ makes all three solvers choke when the starting condition is an equal distance to those solutions in constraint space:
[image]
Here I put a point on both the X axis and on the circle, then removed the X axis constraint and added a Y axis constraint. Putting it at x != 0 has the solvers converge on a (non unique, as pointed out by @petaflot) solution.

Roy-043, 2025-04-22:
The original issue is based on a wrong assumption:

https://wiki.freecad.org/Sketcher_Workbench#Constraints:
> A complex Sketch with multiple objects may have a number of different solutions, and making it fully-constrained can mean that just one of these possible solutions has been reached based on the applied constraints.

@ElementW This is really a different issue.

petaflot, 2025-04-22:
> https://wiki.freecad.org/Sketcher_Workbench#Constraints:

I get 502, so I cannot read the latest content, archive.org gives me an idea.

> A complex Sketch with multiple objects may have a number of different solutions, and making it fully-constrained can mean that just one of these possible solutions has been reached based on the applied constraints.£

we're not talking about a complex sketch here, rather one on the most basic use cases : "fully constrained" IS  NOT "there are at least one solutions, we'll keep the most obvious". also - and this is probably in the same line - specifying a negative length (ie. in reference to an axis, typically when copying objects) always fails, but should put the constraint on the "other side" of the axis. this **needs** to be fixed (I mentioned quaternions for a reason)

Roy-043, 2025-04-22:
I'll edit the Wiki page. This may be hard to accept, but your is assumption is still wrong IMO. Just search the forum for the word "flipping".

petaflot, 2025-04-22:
let's take an assumption and get to the semantics: *fully constrained* is not the same as *fully constrained within the boundaries of flipping that must be searched across the content of a forum (not even the official docs?!?)*

*fully constrained* should mean what it states, as in ***mathematically** fully constrained*. cut this is what we're dealing with :  mathematics and physical objects

petaflot, 2025-04-22:
> This issue is a UX issue first and foremost. Past that it also questions whether the underlying design & technical cause is even considered right in the first place.

I have not looked at the source code, so I don't know how FreeCAD works internally... TBH I've run into singularities on some unusual geometry even with SolidWorks, similar issues with Inkscape ad FreeCAD too... AFAIK none of these use quaternions internally so they run into the same singularity problems : therefore it's not an UX problem, rather related to some deep internal workings - that, if the proper abstraction is done - should not be too hard to fix (although it definitely will be the source of a quite large commit and might require a lot of changes, maybe in the UX too)

ElementW, 2025-04-22:
Then if you mean it this way I believe the issue I've showcased is indeed the same, just another facet of it; fixing your issue would mean either:
* changing the amount/semantics of constraints needed to reach "Fully constrained" so mathematically they're not ambiguous any more, thereby fixing my case
* changing the solvers to be able to disambiguate what to converge to, a proper implementation of which will have to handle cases where constraint system solutions are an equal distance apart (or risk not fixing analogues of your setup where 3+ equidistant solutions exist), _also_ fixing my case

petaflot, 2025-05-01:
> I.e. on the mathematically underconstrained coin, yours is the "it doesn't make sense" side, mine is the practical consequence side: constraint flipping & outright solver failure.

AFAIK mine is "the solver is explicitly biased towards one valid solution, and does not realize there is at least one another", yours is "the solver is exactly between two valid points and doesn't know which to chose, because it is not aware that there are more than one valid solutions"

> specifying a negative length (ie. in reference to an axis, typically when copying objects) always fails

this obviously means the solver is somehow using absolute distances, with no regard to possible solutions in the negative range. If I mentioned **quaternions**, it is because it is the **only** way to have a fully constrained system (think of the gimball lock issue) regardless of null values (avoids ZeroDivisionError)

PaddleStroke, 2025-09-25:
[image]

Someone reported Catia V5 is also reporting fully constrained.

> Should FreeCAD finally become an actual tool, or should it remain a toy like those found in the Kinder Überaschung eggs?`

@petaflot can you please kindly let us know which CAD package is not a toy like those found in kinders? Because apparantly Catia V5 is a toy.