FreeCAD/FreeCAD #20899: Compilation Broken on Cmake version 3.28 and newer

hhassey, 2025-04-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

apparently starting from Cmake version > 3.28  the function The **exec_program()** command should not be called. it has been deprecated. the command **execute_process()** should be called instead.

I have reported and discussed this with more information in the [FreeCAD forum here](https://forum.freecad.org/viewtopic.php?t=96220)

This breaks compilation on Arch Linux wich at the moment of writing uses Cmake 4.0.1

### Full version info

```shell
No freeCad dialog available as I can not compile.
```

### Subproject(s) affected?

None

### Anything else?

I have reported and discussed this with more information in the [FreeCAD forum here](https://forum.freecad.org/viewtopic.php?t=96220)

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

oursland, 2025-04-21:
Hi @hhassey,

Unfortunately some of these issues are in dependencies making it hard to correct them until they are corrected upstream.

If you pass the command line argument `-DCMAKE_POLICY_VERSION_MINIMUM=3.5` to CMake, it will permit you to build.

hhassey, 2025-04-21:
Thanks, That did work. I will mark this solved.

closed 2025-04-21 by hhassey