FreeCAD/FreeCAD #20900: A few issues with MayaGesture

celdorwow, 2025-04-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I found some issues with MayaGestures in FreeCAD v1.0.0.

### Navigation

#### Orientation

First of all, holding <kbd>ALT</kbd> in Maya triggers a general Viewport changes. particularly, <kbd>ALT + LMB</kbd> changes orientation, which works in **FreeCAD** set to _MayaGesture_. However, when I enter _Sketch_ mode, the gesture somewhat alternates between _Selection_ and _Navigation_, even though <kbd>ALT</kbd> is still held.

[Link to a video](https://drive.google.com/file/d/1cXuNSWxIaYBjSU4ZTVoonAJO2bnQQ64f/view?usp=sharing)

#### Zoom

Additionally to vertical gesture, zoom in Maya is also triggered by horizontal gestures while holding <kbd>ALT + RMB</kbd>, which does not work in **FreeCAD**. In Maya, dragging to the left triggers Zoom Out while dragging to the right triggers Zoom In. IMO this could be implemented in order to fully conform with gestures in **Maya**.

Lastly, **Maya** has inverted zooming when draggin mouse vertically. Hence, dragging mouse up (pushing) while holding <kbd>ALT + RMB</kbd> triggers Zoom Out, not Zoom it is inverted in **FreeCAD**. I know we can turn _Invert zoom_ off in _Preferences_, but then this setting also affects zooming on rolling MMB.

### Rotation Focus

For some reason in **FreeCAD**, MayaGesture also adds _Rotation focus_ triggered by both MMB and <kbd>H</kbd>. In _Sketch_ mode, however, this is in conflict with _Constraint Horizontal_ and IMO should be disabled.

[Link to a video](https://drive.google.com/file/d/1RyQV_BWuVeP4CzKPeHe4Dcg1F-miu9I1/view?usp=sharing)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-22:
@Rexbas do you mind weighing-in ?

Rexbas, 2025-04-22:
I'll have a look. If the default zoom behavior in Maya is the opposite then we should probably do that as well.

@celdorwow, how exactly does that horizontal zoom work? What happens when you move the mouse at an 45 degree angle between horizontal/vertical? Can you show video?

celdorwow, 2025-04-23:
> [@celdorwow](https://github.com/celdorwow), how exactly does that horizontal zoom work? What happens when you move the mouse at an 45 degree angle between horizontal/vertical? Can you show video?

Here is a [video](https://drive.google.com/file/d/1Ypz74V7pQgp89suFyyCghm9hKow5edMP/view?usp=sharing), where I demonstrate horizontal, vertical and diagonal zooming.

I guess horizontal zooming simply works along with vertical one. As such, in Maya if I move a cursor either towards left hand-side upper corner or towards right hand-side lower corner the zooming is slightly amplified with former causing zoom out and latter casing zoom in. The two other diagonal gestures reduce the effect. If I move mouse perfectly along 45 degree, the effect is cancelled. All this is included in the linked video.

Hope that helps.
Thanks

luzpaz, 2025-04-24:
Thanks @Rexbas !

closed 2025-04-28 by chennes

celdorwow, 2025-04-28:
@yorikvanhavre Thanks. I tried to compile in MSYSG32-MING64 following instructions but i've got an error, something around these lines:

    Cannot find correct cmake configuration for yaml-cpp ...

I don't know what I am doing so I will wait for binaries. Cheers!

celdorwow, 2025-05-01:
> The PR is now merged, [@celdorwow](https://github.com/celdorwow) please test whenever you can, with the next weekly build!

Hi @yorikvanhavre.

I have tested the newest weekly build

[FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311.7z](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311.7z)

I am afraid zooming is still inverted to what it is in Maya. Also, moving mouse left and right seems to cancel zooming effect in FreeCAD, while in Maya the diagonal gesture does so.

In Maya the <kbd>ALT + RMB</kbd> and moving mouse left causes Zoom Out, so does moving mouse up. And then both gestures moving right and pulling down cause Zoom In (see the first part of the [video](https://drive.google.com/file/d/1Ypz74V7pQgp89suFyyCghm9hKow5edMP/view?usp=sharing).)

**Other issues**

I mentioned other issues in the first comment. Perhaps I should open a new ticket for those. Please let me know.

When `MayaGestures` is on, navigation in `Sketcher` mode (`Part Design`) is affected because <kbd>H</kbd> is set to centre focus items in Viewport  by `MayaGesture`. This is in conflict with shortcut to horizontal constraint, which is also <kbd>H</kbd>.

<kbd>ALT + LMB</kbd> alternates between two effects selection and rotation, whereas it's supposed to only rotate Viewports.

celdorwow, 2025-05-02:
Hi @Rexbas,

I can't open FreeCAD with the same issue as in #21052.

celdorwow, 2025-05-08:
> The PR is now merged, [@celdorwow](https://github.com/celdorwow) please test whenever you can, with the next weekly build!

Hi @yorikvanhavre ,

I double checked MayaGesture in the [build weekly-2025.05.06](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.05.06) and navigation is correct.

The other two issues are also fixed: those related to rotation in Sketcher and the focus key (<kbd>H</kbd>).

Thanks

PR #20948 (2025-04-28): Gui: MayaGesture navigation fixes
Fixes #20900.

Note: the zooming behavior changes are based on the description and video by @celdorwow. I don´t have Maya so I cannot confirm that this is indeed the default behavior in Maya.