FreeCAD/FreeCAD #20907: RFE: Spreadsheed - 'omit instant recompute' | 'recompute now'

herrdeh, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

My most used models takes ~30s to recalculate, when a parameter in Spreadsheet is changed. It has from 6 to 12 parameters, which need to be changed when I adapt it to a new usecase, which I need to do several times a week. Especially annoying is when a text line needs to be changed, this takes the same recompute time.

Could we have 
* a selector "omit instant recompute" at the spreadsheed - and
* a button "recompute now" ?

Thus, I could punch in all my parameters, and then go for a cup of tea...
I could offer 100€ for this, hopefully some fellow joins in.

Cheers,
Wolf

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.12
  * dodo 1.0.1
  * ExtremeProLight 1.6.0
  * Curves 0.6.29
  * BOLTSFC
  * nurbs
  * animation
  * IconThemes
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pitch-circle, 2025-04-23:
>     * a selector "omit instant recompute" at the spreadsheet - and
>     * a button "recompute now" ?

Please could you clarify that you want this to apply only to the spreadsheet.

In the tree view, if you right click the document, you can choose 'skip recomputes'.

There is also StdToggleFreeze (https://wiki.freecad.org/Std_ToggleFreeze).

There is also the 'allow partial recomputes' whose exact functionality seems unknown (https://forum.freecad.org/viewtopic.php?t=48052).

The F5 keyboard shortcut is a good proxy for 'recompute now'. There is also a Macro ForceRecompute which is relevant (https://wiki.freecad.org/Std_Refresh)

It may be good to say how the proposed feature fits in with or extends what is currently provided.

herrdeh, 2025-05-05:
> In the tree view, if you right click the document, you can choose 'skip recomputes'.

When thinking about it, this is the much better option. Only problem is that I could not find a way to toggle it on/off with a keyboard shortcut / menu entry. I'd use it so extensively that the right-click switch is a little bit of a pain...

So if there is no method for such a shortcut, I'd close this ticket and open one for this shortcut.

Thanks a lot in advance,
W

herrdeh, 2025-05-11:
Closed, because new #21246 provides a much better solution.

closed 2025-05-11 by herrdeh