FreeCAD/FreeCAD #20916: Unable to change constraint value - popup not opening

Anntoin, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi,

I'm new to FreeCAD but this seems like a bug. When trying to set a value for a constraint I'm unable to click on 'Change value' from the RMB menu on the constraints panel. I _can_ click 'Change value' in the RMB menu over the constraint in the sketch view but the popup to change the value doesn't appear. I'm encountering this issue after starting the [basic part design tutorial](https://wiki.freecad.org/Basic_Part_Design_Tutorial_019). 

Have tried using the Arch packaged version, 1.0 AppImage, and the weekly AppImage. I'm using Sway/Wayland and have tried with `QT_QPA_PLATFORM=xcb` & `QT_QPA_PLATFORM=wayland` without any change. I'm also getting the same behaviour in 'safe mode'

Interestingly in the weekly AppImage 'Change value' and 'Toggle to/from reference' are greyed out (rather than just not hightlighted) but 'Change value' on sketch view still does nothing.

When I'm able to click on 'Change value' the python console shows `>>> # Gui.runCommand('Sketcher_ChangeDimensionConstraint',0)`, but I don't get anything more than that.

Menus I'm referring to:
[image]

[image]

Let me know if there's any other details I can provide

### Full version info

```shell
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-04-22:
You are dealing with a horizontal constraint here (which does not have a value). You have given it a very awkward name though.

I am closing this as a user error.

Anntoin, 2025-04-22:
Think I know where I'm going wrong. 'Change value' shouldn't apply to this type of constraint (horizontal & vertical) and I should use a dimension constraint (which does work).

In that case the only bug here is that 'Change value' shouldn't be clickable from the RMB menu on the sketch view.

I'm guessing the horizontal/vertical constraint behaviour changed at some point and the tutorial is out of date.

Roy-043, 2025-04-22:
You are not following the tutorial at all BTW. Please pay more attention.

Roy-043, 2025-04-22:
Sigh...

Anntoin, 2025-04-22:
I appreciate that I had mixed up the constraint types and that sent me down a wrong rabbit hole.

However, the (admittedly minor but still confusing) UI issue in the sketch view still holds. The UI issue in the constraint panel having been fixed in the weekly build.

For a horizontal constraint you can click 'Change value' which shows as the command 'Sketcher_ChangeDimensionConstraint' in the python console. 'Change value should either be greyed out or simply not present on this menu. When clicked there is no error message and to an end user this just makes it look like the UI is unresponsive rather than you have tried something that shouldn't be possible.

Roy-043, 2025-04-22:
New issue created: #20917.