FreeCAD/FreeCAD #20919: Recompute works unreliably when entire tree is marked

violettinged, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a file where some feature that previously failed has now been fixed. The only thing missing is a recompute.
However, right click on the entire tree and selecting 'Mark to recompute' followed by a recompute does not work as expected
(sometimes it finally works after 2 repetitions, sometimes it still doesn't do anything even after 10 repetitions):

[image]

In contrast, restarting FreeCAD and only recomputing the feature (almost always) works as expected for me:

[image]

I also tested 1.1.0dev.41314 (Architecture: x86_64, not arm64!), the problem is also present there. Please find attached the FCStd file (just remove that .zip extension from the file). I couldn't find an existing issue covering this problem.

[recompute_fails.FCStd.zip](https://github.com/user-attachments/files/19852565/recompute_fails.FCStd.zip)

### Full version info

```shell
OS: macOS 14.7.4
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xiphmont, 2025-04-23:
The example above looks very much like a precision/tolerance problem, and it is currently one of the banes of using FreeCAD.

FreeCAD's boolean operations (including fusions built into other operations) usually decide coincident/angle/intersection tolerances automatically, often unexpectedly coarse or unexpectedly tight, and will silently apply them to the operation in progress as well as the inputs (sometimes).  Tolerances only ever get coarser, except during a forced recompute.  In models that use inputs multiple times across a part, these tolerances will spread forward/backward through the model in unexpected ways, it's mostly invisible, and the tolerance information is cached.  

This often causes booleans, fuses, refines, etc, to randomly fail... or suddenly work again... during a full-tree recomputation, where all the tolerances are force-set set back to a starting value of Precision::Confusion() (that is, 1e-7) and everything is recomputed anew.

If the model was 'working' and a full-tree recomputation 'breaks' it, it was already broken but nothing exposed the breakage.  Automatic tolerance changes do not trigger automatic recomputation, even when they affect the result of computing the object.

You can check that partial and full recomputations are not beginning from the same starting tolerances during that object's computation. The Curves workbench exposes an 'information' mode button (white i in a blue circle) that shows the tolerance tied to any selected vertex/line/plane.  You can use this to see what the tolerances of your objects are before and after recompute.

xiphmont, 2025-04-24:
[Been thinking about this more, because I'm trying to learn this aspect of the code now...]
BTW, take the exact details of what I said with a grain of salt/skepticism.  One thing that's clear to me right now is that I have a broad handle on the issue, but I don't yet really understand it.  I know it bites me too (hard) on a regular basis, but if I really understood the problem, I could try to fix it.  So it's pretty clear I don't yet really understand enough of the problem.

maxwxyz, 2025-04-25:
Could it be because of the boolean operation?

FEA-eng, 2025-10-07:
Still relevant

In fact, this is pretty bad because when checking the issues from previous versions reported here, we rely on recompute heavily.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```