FreeCAD/FreeCAD #20922: Errors when starting a new CAM Job

JohnOCFII, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Report window fills with error message.  Seems core message is related to an unsupported character...

This error does not occur in 1.0.0 nor as recently as weekly build 40444.

Steps to recreate:

1) Open attached model (created in PartDesign)
2) Switch to CAM workbench
3) Click the icon to create a new CAM Job
4) In the "Create Job" dialog window, confirm the one Body (TestBody) is selected and hit `OK`

The issue seems to based on the following, but I was not successful in narrowing it down to where, exactly it originated, but it has something to do with Post Processing.

If you create a CAM job in a previous version, and then attempt to edit it in the current weekly, this message does not occur.  Seems to only happen on new models.

Check the Report Window to see the following:

```
13:37:01  pyException: Traceback (most recent call last):
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/Mod/CAM/Path/Main/Job.py", line 524, in onChanged
    processor = PostProcessorFactory.get_post_processor(obj, obj.PostProcessor)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/Mod/CAM/Path/Post/Processor.py", line 104, in get_post_processor
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/Mod/CAM/Path/Post/scripts/KineticNCBeamicon2_post.py", line 87, in <module>
    TOOLTIP_ARGS = parser.format_help()
                   ^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/lib/python3.11/argparse.py", line 2595, in format_help
    return formatter.format_help()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/lib/python3.11/argparse.py", line 287, in format_help
    help = self._root_section.format_help()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/lib/python3.11/argparse.py", line 217, in format_help
    item_help = join([func(*args) for func, args in self.items])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/lib/python3.11/argparse.py", line 217, in <listcomp>
    item_help = join([func(*args) for func, args in self.items])
                      ^^^^^^^^^^^
  File "/Applications/Other Apps/FreeCAD-41354.app/Contents/Resources/lib/python3.11/argparse.py", line 217, in format_help
    item_help = join([func(*args) for func, args in self.items])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/O<class 'ValueError'>: unsupported format character '\' (0x5c) at index 62
```

### Full version info

```shell
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41354 (Git)
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

CAM

### Anything else?

[CircleBoxTest.FCStd.zip](https://github.com/user-attachments/files/19855340/CircleBoxTest.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-04-22:
Can not reproduce
Please show arguments for postprocessor

[image]

knipknap, 2025-04-22:
This would probably be fixed by #20877

JohnOCFII, 2025-04-22:
This is a fresh run with all old preferences removed, but the fact that x40444 didn't display the errors, and the more recent FreeCAD versions do, likely points to recent changes around the post processor, that, perhaps, only effect configurations that haven't previously been used with the CAM workbench. (Just a guess).

> Please show arguments and selected postprocessor

[image]

maxwxyz, 2025-04-23:
@knipknap is this fixed by #20877?

closed 2025-05-05 by sliptonic