FreeCAD/FreeCAD #20923: Status Bar - Missing

bstott, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Status Bar information: notifications, mouse, dimensions, zoom - disappeared.

Within the last two (?) weekly rebuilds - The Status Bar is empty.  I have de-selected/selected View>Status Bar. Restarted FreeCAD and the Status bar only shows Open File Tabs.   How do I get the mouse, notifications, zoom, etc back?  I have checked other FreeCAD versions - the Status Bar is there.

**My Machine Info:**
OS: Windows 10 build 19041
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
Installed mods: 
  * 3D_Printing_Tools
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * Defeaturing 1.2.3
  * Design456 0.0.1
  * fcVM 2024.9.5
  * freecad.gears 1.3.0
  * Launcher
  * OpticsWorkbench 1.1.2
  * PieMenu 1.11.3
  * Silk 0.1.5

### Full version info

```shell
OS: Windows 10 build 19041
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
Installed mods: 
  * 3D_Printing_Tools
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * Defeaturing 1.2.3
  * Design456 0.0.1
  * fcVM 2024.9.5
  * freecad.gears 1.3.0
  * Launcher
  * OpticsWorkbench 1.1.2
  * PieMenu 1.11.3
  * Silk 0.1.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

adrianinsaval, 2025-04-23:
can you reproduce with a clean config? (rename or remove %APPDATA%/FreeCAD folder) I can't reproduce this issue

bstott, 2025-04-23:
The Status Bar information: notification, mouse,  and dimensions disappear off bottom of the screen when I maximize the application window.  The status info is available if I manually size the window to fit the monitor.    The application is otherwise working.  I am not going to delete the configuration file, yet.  I'm done today. I'll look at it tomorrow.

xtemp09, 2025-04-23:
@bstott, could you upload the screenshots? It is possible that I experience this too, but it results from buggy video drivers (or something else) and observed, obviously, with all programs except for _MS Office_ and _Notepad++_.

bstott, 2025-04-23:
04.23.2025 -  Please close.    I restarted my laptop and the anomaly isn't happening.   I don't know the cause. It clearly isn't FreeCAD related.

closed 2025-04-23 by bstott