FreeCAD/FreeCAD #20926: CAM: Stock from ExistingStock moves in respect to parts

otwieracz, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After selecting existing solid as a stock in CAM job, the stock moves in relation to the machined bodies.

Not much to explain, video will convey it better:  https://youtu.be/ymJvZo6V4Bs
ZIP archive with test files where I isolated the test case: [Archive.zip](https://github.com/user-attachments/files/19857698/Archive.zip)

If by any chance this turns out to be a low hanging fruit/good first issue to work on, I'll be more than happy to participate if you were able to guide me - I would really like to try contributing to FreeCAD.

### Full version info

```shell
OS: macOS 15.4.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
  * fasteners 0.5.30
  * ExtremeProDark 2.8.0
  * ExtremeProLight 1.6.0
  * Curves 0.6.9
  * Assembly3 0.12.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

otwieracz, 2025-04-25:
@phaseloop I only caused confusion with this visibility toggle (and I can't start FreeCAD CAM now due to different issue with numpy on macOS).

**The issue is presented in the very last moments of the video, where after re-opening the job the clone of a stock moves in respect to the machined bodies, changing the setup completely**. 

**Before reopen**

[image]

**After reopen** stock moves in relation to CAM cloned bodies.

[image]

luzpaz, 2025-05-10:
> (and I can't start FreeCAD CAM now due to different issue with numpy on macOS)

An aside, @otwieracz is this the numpy bug ? 
* https://github.com/FreeCAD/FreeCAD-Bundle/issues/375

otwieracz, 2025-05-11:
@luzpaz Yes, I was able to reproduce it on 1.1dev (`master`, actually, built from source).

I'm still trying to debug what is going on exactly, but what I discovered so far is that the `Stock-Stock` clone in `Job` is being recreated every time the job properties panel is open here:
https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/Mod/CAM/Path/Main/Gui/Job.py#L625

and (not 100% yet) it might be caused by the callback:
https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/Mod/CAM/Path/Main/Gui/Job.py#L662

is firing when calling the `setFields`:
https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/Mod/CAM/Path/Main/Gui/Job.py#L650

-- 
Regarding problems with startup: It was a `numpy` import issue, but I managed to work around it with: 'https://raw.githubusercontent.com/imanolpg/workaround-dyld-duplicate-rpath-freecad/refs/heads/main/workaround-dyld-duplicate-rpath-freecad'

otwieracz, 2025-05-11:
I think I see what is going on, will try to create PR soon.

luzpaz, 2025-05-11:
@otwieracz nice, please also post the revision of HEAD that you tested 1.1dev on, TIA! 

> will try to create PR soon.

That's would be amazing, TY!

luzpaz, 2025-05-12:
Thank you for your efforts, @otwieracz 👍

PR #21270 (2025-05-13): CAM: Fix Existing Stock from resetting
Due to a Qt signal setup, the clone object for Existing Stock stock type was recreated every time the Job properties dialog was opened (during the stock candidates list population).

This fix blocks the Qt signal from being emitted during the dropdown population.

## Issues

- Fixes #20926
closed 2025-05-13 by kadet1090